ms-iot/samples #1: Python libraries wingpio, winspi, wini2c not yet published?

tdicola, 2015-05-02:
Sorry if this is the wrong spot to raise an issue like this, but I'm hoping it can make it to the right folks.  I'm checking out the Python IoT preview samples (PythonBlinky, PythonBlinkyServer) but I can't seem to find the source for the wingpio, winspi, and wini2c libraries that provide Python access to GPIO, SPi, I2C, etc.  The readme.txt for the PTVS IoT preview mentions a repository that should be here:
```
To use GPIO, I2C or SPI from python project:
=============================================
Option 1 - Use source projects for extensions (useful in case you are building same app for multiple platforms)
=============================================
1. Enlist in http://github.com/ms-iot/python
2. Pull down entire project
3. Add projects (e.g. wingpio, winspi or wini2c) to your python solution
...
```
However it looks like http://github.com/ms-iot/python doesn't exist or is a private repo right now.  If possible it would be great to make this available for folks to check out.

If the source isn't ready to be published yet, are there any docs that describe how to use those python libraries or their interface?  

I'm curious to check them out because I create and maintain python modules for Adafruit that abstract away access to GPIO, I2C, SPI, etc. across multiple platforms and am curious about adding support for the new Windows IoT core system.  The library ([here](https://github.com/adafruit/Adafruit_Python_GPIO)) hides platform-specific hardware access behind a standard interface that other modules can code against and then easily run on various platforms (like the Raspberry Pi and Beaglebone Black).  If I can add support for the Windows IoT core python libraries then a lot of fun hardware could easily work on win IoT, like [TFT](https://learn.adafruit.com/user-space-spi-tft-python-library-ili9341-2-8/overview) and [OLED displays] (https://learn.adafruit.com/ssd1306-oled-displays-with-raspberry-pi-and-beaglebone-black/overview), [LED matrices](https://learn.adafruit.com/led-backpack-displays-on-raspberry-pi-and-beaglebone-black), [temperature sensors](https://blog.adafruit.com/2014/09/19/adafruit-temperature-sensor-guides-for-raspberry-pi-beaglebone-black/), [touch sensors](https://learn.adafruit.com/mpr121-capacitive-touch-sensor-on-raspberry-pi-and-beaglebone-black), [NFC readers](https://learn.adafruit.com/raspberry-pi-nfc-minecraft-blocks) and [more](https://github.com/adafruit?query=adafruit_python).

Let me know if it's possible to share the Python windevices library code or at least interfaces.  Thanks!

GQHD168, 2015-05-02:
I'm sorry about that. I just start learning. Sorry.

发自我的 Windows Phone

 /09/19/a
 dafruit-temperature-sensor-guides-for-raspberry-pi-beaglebone-black/), [touch sensors](https://learn.adafruit.com/mpr121-capacitive-touch-sensor-on-raspberry-pi-and-beaglebone-black), [NFC readers](https://learn.adafruit.com/raspberry-pi-nfc-minecraft-blocks) and [more](https://github.com/adafruit?query=adafruit_python).

https://github.com/ms-iot/samples/issues/1

alecont, 2015-05-05:
this is the right place! thx for reporting this, tdicola. We messed up a little our links, and we're correcting this now.

Also, we'd love to collaborate to get your libs to support Win10 IoT Core! And get your input on our versions of the libs.

munyirik, 2015-08-05:
Hi @tdicola, you may know this already but just in case, http://github.com/ms-iot/python is public now. And the code for wingpio, winspi, and wini2c is shared there.

closed 2015-08-05 by bfjelds