ms-iot/samples #2: Cinder or openFrameworks sample?

brucelane, 2015-05-02:
hello,
It would be great to have a sample for this frameworks.
Thank you

ooeygui, 2015-05-03:
We would love to have some help on this.

alecont, 2015-05-05:
Yep! I love Cinder and OF from my days in VC++ :-) MS OpenTech worked a bunch on Cinder and OF for ports to Windows Store (which will be beneficial for the UWP angle).
One of the main issues I can see is that we don't have yet a performant DirectX stack on the Rpi2. But it's there (already) on the MBM.

I'm checking with some of the folks that worked on Cinder and OF, and I'll report back.

@brucelane: what projects/samples do you have in mind?

brucelane, 2015-05-05:
@alecont : Andrew Bell from the Cinder team is working on the glNext branch which can use ANGLE but neeeds more time to add the ARM branch of it... 
I'm a visual artist building my own VJ software, and I would like to use Cinder on the RPI2 on W10 to use it as a MIDI (or other devices) network router so send WebSocket messages.
It would imply maybe a node.js server (which is working fine right now)
Also a openGL fragment shader browser and/or editor for live-coding.
My goal is to create a remote UI for my software.

GQHD168, 2015-05-05:
Hi，i'm using gpio of rpi2，but i can't find the interface of gpio in vs 2015rc, no matter I use c++ or c#.  What's the problem?

发自我的 Windows Phone

https://github.com/ms-iot/samples/issues/2

olivierbloch, 2015-05-05:
Hi @GQHD168 
Chances are this is due to the fact that the original commit was referencing an older version of the SDK. I just merged a pull request that solves this issue.
Let me know if that works for you.

closed 2015-05-11 by ooeygui