ms-iot/samples #6: AllJoyn Adapter Host App - Error a value of type "BridgeRT::IAdapterValue ^" cannot be used to initialize an entity of type "AdapterLib::AdapterValue ^"	AdapterLib

plkumar, 2015-05-06:
Severity	Code	Description	Project	File	Line
Error	C2440	'initializing': cannot convert from 'BridgeRT::IAdapterValue ^' to 'AdapterLib::AdapterValue ^'	AdapterLib	D:\Projects\Windows-universal-samples\SampleAllJoynApp\SampleAllJoynApp\AdapterLib\Adapter.cpp	172

Severity	Code	Description	Project	File	Line
Error		IntelliSense: a value of type "BridgeRT::IAdapterValue ^" cannot be used to initialize an entity of type "AdapterLib::AdapterValue ^"	AdapterLib	d:\Projects\Windows-universal-samples\SampleAllJoynApp\SampleAllJoynApp\AdapterLib\Adapter.cpp	172

charlehsin, 2015-05-06:
My workaround is:
AdapterValue^ attribute = ref new AdapterValue("");
attribute->Set(adapterProperty->Attributes->GetAt(0));

zhuridartem, 2015-05-08:
The document is missing a dynamic_cast<AdapterValue^> for the following line:

AdapterProperty^ adapterProperty = dynamic_cast<AdapterProperty^>(Property); 
AdapterValue^ attribute = adapterProperty->Attributes->GetAt(0);

should be:

AdapterProperty^ adapterProperty = dynamic_cast<AdapterProperty^>(Property); 
AdapterValue^ attribute = dynamic_cast<AdapterValue^>(adapterProperty->Attributes->GetAt(0));

zhuridartem, 2015-05-11:
Documentation at https://github.com/ms-iot/samples/blob/develop/AllJoyn/AllJoynDSBGuide/AllJoyn_DSB_GPIODevice_Sample_Tutorial_v1.0.pdf was updated with correction.

ooeygui, 2015-05-19:
Wrong bug

ooeygui, 2015-05-19:
This one was correctly closed; per conversation with @zhuridartem

closed 2015-05-19 by ooeygui