ms-iot/samples #8: AllJoyn DSB Template: AdapterBackgroundService deploy error.

charlehsin, 2015-05-06:
I am trying to run the AllJoyn DSB template and follow the GpioDevice Sample Tutorial.
I can succesfully run AdapterHostApp on my Win10 PC. However, when I tried to deploy "AdapterBackgroundService" to a MBM Win IoT Core system, I got the following error:

Error : DEP0700 : Registration of the app failed. C:\.......\App1AdapterBackgroundServiceVS.Debug_Win32.Tester\AppxManifest.xml(25,6): error 0x80070490: Cannot register the App1AdapterBackgroundService_1.0.0.0_x86__XXXXX package because the following error was encountered while parsing the App1AdapterBackgroundService application: Element not found. Try again and contact the package publisher if the problem persists. (0x80073cf6).

Can any one help?

zhuridartem, 2015-05-06:
Hello,

We've reproduced this issue. Turns out the required Assets\ that are referenced in project AdapterBackgroundService file Package.appxmanifest are missing. They are present in the AdapterHostApp project.

So, to fix:
1. Copy Assets directory from AdapterHostApp folder to AdapterBackgroundService folder.
2. Under AdapterBackgroundService project, right click, Add -> New Filter, call it "Assets" 
2. Right click on the created "Assets" filter,  Add -> Existing Item, and select all files under AdapterBackgroundService\Assets\

This will add all the missing Assets files back to AdapterBackgroundService.

charlehsin, 2015-05-07:
Great. It fixes it.  By the way, do you know why there can only be 1 background service running at 1 time?  When I deploy this, the in-box ZWave Adapter service stops.

closed 2015-05-07 by charlehsin

zhuridartem, 2015-05-11:
The template DeviceSystemBridgeTemplate.vsix file at https://github.com/ms-iot/samples/tree/develop/AllJoyn/AllJoynDSBGuide has been updated with the Assets fix.