ms-iot/samples #10: Only 1 AllJoyn background service can run on Win IoT at 1 time?

charlehsin, 2015-05-07:
Originally, the in-box ZWaveBackgroundService is running. However, when I deployed the dummy AllJoyn AdapterBackgroundSerivce (created from DSBTemplate), ZWaveBackgroundService stoped automatically. I would imagine that multiple instances should be allowed to run, to support multiple devices. Can any one help  on this issue?
Chih-fan

zhuridartem, 2015-05-08:
We also noticed that existing DSB Background Service gets stopped when deploying another one from Visual Studio. Once deployed from Visual Studio, you can use iotstartup.exe to add the new BackgroundService as a headless startup task so it starts on next boot.

Using remote powershell session (http://ms-iot.github.io/content/win10/samples/PowerShell.htm), run "IotStartup add headless AdapterBackgroundService"

Then run "IotStartup list headless" to make sure both ZWaveBackgroundService and AdapterBackgroundService are registered.

Reboot and both should get started now.

For more details on setting startup app, see http://ms-iot.github.io/content/win10/tools/CommandLineUtils.htm

charlehsin, 2015-05-08:
Thanks.... That works.

closed 2015-05-08 by charlehsin