ms-iot/samples #11: MessageDialog not working.

parnz, 2015-05-08:
Hello Developers, I'm trying to display a message box in rpi2 I take the code from [THIS](https://github.com/Microsoft/Windows-universal-samples/tree/master/messagedialog) (C#) and deploy it on rpi2 running windows 10 iot but it does not works. But it works on my pc (windows 10 pro insider preview build 10074 64bit + visual studio 2015 rc).

If this code not compatible with windows iot core then provide me a sample code please.
I'm planing to write an app in headed mode.

Thank you.
Wisitchai.

endy66, 2015-05-12:
Same Problem here. If i deploy the code to win10 preview pc, the MessageDialog works perfect. On Win IoT Core (RPI2), the MessageDialog does not appear.

ooeygui, 2015-05-12:
Hello, Thank you for the report. On the desktop, the Windows Shell provides dialog boxes and other user interface overlays. Windows 10 Core does not have a shell component, so the Maker's application needs to provide all user experience elements.

closed 2015-05-12 by ooeygui

parnz, 2015-05-14:
>**the Maker's application needs to provide all user experience elements.**

That means I have to create my custom MessageBox class myself right?

Thanks.

mms-, 2015-11-25:
That makes sense, however there should be a natural way to implement the dialog box service, there shouldn't be an API that exists but crashes the app..

That is, is there a way to plugin our own dialog functionality without going in and wrapping dialog code all over the app?