ms-iot/samples #12: AllJoyn DSB: BridgeRT codes

charlehsin, 2015-05-08:
I am sorry about this post. This is not really an "issue", but a question. However, I don't know where else to post this question. The question is:
I am going through BridgeRT codes that come with DSB template. What are the relations between these codes and Windows.Devices.AllJoyn namespace APIs and the original AllJoyn framework SDK for Windows?
Are BridgeRT codes using AllJoyn framework SDK? 
Since "searching in codes for Windows.Devices.AllJoyn" does not return any result, BridgeRT is probably not using Windows .Devices.AllJoyn APIs?
Are there some document about the overview design and interfaces and architecture of BridgeRT?
Thanks,
Chih-fan Hsin

zhuridartem, 2015-05-11:
Overview of the BridgeRT at https://git.allseenalliance.org/cgit/dsb.git/plain/Docs/AllJoyn%20-%20Device%20System%20Bridge%20-%20Whitepaper%20v1.0.pdf

Overview of AllJoyn API surface for UWPs in Windows 10 at https://channel9.msdn.com/Blogs/Internet-of-Things-Blog/Step-By-Step-Building-AllJoyn-Universal-Windows-Apps-for-Windows-10-Public-Preview

Both the BridgeRT and Windows.Devices.AllJoyn use MSAJAPI.dll, which is AllJoyn Standard Client C API with reference at https://allseenalliance.org/docs/api/c/index.html

charlehsin, 2015-05-11:
Thanks for the reply. I have another question related to this.
I understand the idea of DSB. However, in my understanding, one can also use Windows.Devlices.AllJoyn to create a AllJoyn Producer and use the target device's SDK to hook up the device and the AllJoyn network. Why do we need a DSB, instead of using AllJoyn Producer? Does DSB make the whole process easier, or/and does DSB provide other features/capabilities?
Thanks,
Chih-fan

zhuridartem, 2015-05-12:
The primary targets for the Device System Bridge are existing device networks. Our goal with DSB is to offer a cost efficient and quick method to bring existing devices to AllJoyn as virtual devices on the AllJoyn network through a pre-baked AllJoyn producer implementation (Bridge), and implementing calls to SDK hooks to support translating between AllJoyn concepts to target devices (Adapter). As a developer you don’t need to touch AllJoyn producer with a DSB, unless want to change default behavior of the Bridge.

For existing devices, you should go with DSB to allow those devices to be visible and controllable from AllJoyn bus. 
The bridge generates AllJoyn services, interfaces, objects, properties, methods and signals based on device data and capabilities returned from the Adapter. This allows the DSB to handle a wide variety of devices such as ZWave or BACnet from the same reference code provided in the Visual Studio template.

For new devices, you should go with AllJoyn APIs (such as Windows.Devices.AllJoyn) to create an AllJoyn producer.
Note that there is also an AllJoynCodeGen Tool (https://msdn.microsoft.com/en-us/library/dn913809.aspx) that generates Windows.Devices.AllJoyn producers or consumer code to make it easier. You will need to declare the introspect upfront.

There is also a useful tool to generate AllJoyn introspect XML from an existing AllJoyn device: https://github.com/MS-brock/AllJoynToasterDemo/tree/master/getajxml, usage described in detail on https://channel9.msdn.com/Blogs/Internet-of-Things-Blog/Step-By-Step-Building-AllJoyn-Universal-Windows-Apps-for-Windows-10-Public-Preview

closed 2015-05-12 by charlehsin