ms-iot/samples #14: AllJoyn DSB:  OpenZWave library ported to WinRT?

charlehsin, 2015-05-08:
Originally, OpenZWave library is a standard C/C++ library. Did Microsoft port the original OpenZWave library as a Static Library (C++/CX) to be used in UWP (Windows Runtime components)?
I am trying to figure out what tasks need to be done, if one wants to support other network protocol/framework for AllJoyn DSB. For example, if I want to expose 6LoWPAN devices to AllJoyn DSB, do I need to port the 6LoWPAN library to support UWP?

Thanks,
Chih-fan

ooeygui, 2015-05-11:
OpenZWave was ported. The sources are available at the AllSeen git: https://git.allseenalliance.org/cgit/dsb.git.

We are using the DSB to wrap devices that are exposed by non-AllJoyn networks as virtual AllJoyn devices.  For 6LoWPAN the end-points are typically IPv6 addresses which is more of a transport than a virtualizable device.  One option would be to translate your 6LoWPAN ipv6 address into an ipv4 one and just use a standard AllJoyn producer rather than our DSB to create a bus attachment (device) and expose the signals, methods, and properties you want.  AllJoyn will not support ipv6 until later this year.

closed 2015-05-11 by ooeygui

charlehsin, 2015-05-11:
Thanks for the reply. I have another question related to this.
I understand the idea of DSB. However, one can also use Windows.Devlices.AllJoyn to create a AllJoyn Producer and use the target device's SDK to hook up the device and the AllJoyn network. Why do we need a DSB, instead of using AllJoyn Producer?
Thanks,
Chih-fan