ms-iot/samples #17: AllJoyn\ZWaveAdapter\ZWaveAdapterTestApp failed to run.

charlehsin, 2015-05-14:
Hi, I am trying to do a local debug run of ZWaveAdapterTestApp on my Win 10 PC. It failed at Line 376 of ZWaveAdapterTest.cpp file, when it calls 
status = this->adapter->GetConfiguration(&configData);

The error is:
WinRT information: An item cannot be found at the specified path (C:\...........\localState\options.xml)...

Is the TestApp project missing some files?

A more general question: the document available does not talk about this TestApp project. What does this TestApp project do?

Chih-fan

zhuridartem, 2015-05-19:
There are three application projects with the ZWaveAdapter sample.
1. ZWaveBackgroundService - a background service app that initializes bridge and adapter from a startup task. On Windows 10 IOT Core a startup task can be registered for automatic start as a background service type application. Used for deploying and debugging on Windows 10 IOT Core. 

2. AdapterHostApp - simple app that initializes bridge and adapter on UI activation. Used for deploying and debugging on Windows 10 desktop.

3. AdapterTestApp - simple app that was used for testing the implementation of the adapter. It skips the bridge and instantiates the adapter lib directly and exercises entry points of the adapter. Sort of like a Unit Test. You can safely remove this project. And you are correct, options.xml is missing from the project.

ooeygui, 2015-05-19:
Closing per conversation with @zhuridartem

closed 2015-05-19 by ooeygui