ms-iot/samples #21: Unnecessary value assignment in BlinkyWebServerExample - WebServerTask.cs#113

tscholze, 2015-05-31:
### Issue
There is maybe an unnecessary value assignment at the BlinkyWebServerExample.

### Source
```c#
 public HttpServer(int serverPort, AppServiceConnection appServiceConnection)
 {
            listener = new StreamSocketListener();
            port = serverPort; 
            appServiceConnection = appServiceConnection; //little bit confusing
            listener.ConnectionReceived += (s, e) => ProcessRequestAsync(e.Socket);
}
```
### Links
* [Related file on Github](https://github.com/ms-iot/samples/blob/develop/App2App%20WebServer/WebServerApp/WebServerTask/BackgroundTask.cs)
* [Related MSDN forum post](https://social.msdn.microsoft.com/Forums/en-US/3e102a4f-156e-4198-b21c-c87066306712/blinky-web-server-sample-code?forum=WindowsIoT)

jklemmack, 2015-06-01:
It is taking the variable passed in to the constructor and assigning to a private variable of the same name that's a class member.  It could be rewritten as

    private AppServiceConnection appServiceConnection;

    public HttpServer(int serverPort, AppServiceConnection randomVariable)
     {
            listener = new StreamSocketListener();
            port = serverPort; 
            this.appServiceConnection = randomVariable; 
            listener.ConnectionReceived += (s, e) => ProcessRequestAsync(e.Socket);
    }

tscholze, 2015-06-02:
Ah, I see.
I would recommand that little refactoring. It would help beginners to understand the source without the question in mind "why does vs mark this yellow".

bfjelds, 2015-07-02:
Agreed.  We have updated the sample to clarify this.  Thanks!

closed 2015-07-02 by bfjelds

tscholze, 2015-07-05:
No problem. I'm glad I could help. :)