ms-iot/samples #28: Cannot deploy app to Raspberry Pi 2 (win10 IOT Core)

muziekklas, 2015-06-09:
In Visual Studio 2015 RC (Windows 10 Preview) an error occures while deploying:
Error : DEP0800 : The required framework "C:\Program Files\Microsoft SDKs\Windows Kits\10\ExtensionSDKs\Microsoft.NET.CoreRuntime\1.0\.\AppX\ARM\Microsoft.NET.CoreRuntime.1.0.appx" failed to install.

bfjelds, 2015-06-09:
I'm sorry that the samples aren't working for you!  To help us track down the problem and fix it, can you please specify which sample you are seeing this error for?

muziekklas, 2015-06-09:
Blinky (normal one)
I also followed steps on http://ms-iot.github.io/content/en-US/win10/samples/Blinky.htm but mainly used the code from the repo

muziekklas, 2015-06-10:
I fixed it by reflashing the SD card. I assume it was not correctly flashed the first time because I see more Processes on the webUI

closed 2015-06-10 by muziekklas