ms-iot/samples #30: Any camera samples coming soon?

elmerm, 2015-06-21:
I'd like to interact with the Raspberry Pi camera but I don't see anything in the framework for this. Is there something I'm missing? Planning to release any examples soon on this? 

If nothing available, would appreciate some info on how to get the Python library (https://github.com/waveform80/picamera) working with Windows 10. Thanks.

ooeygui, 2015-06-21:
Hi @elmerm,
We do not have drivers planned for the pi camera interface. We will however be supporting USB camera in a coming release. With a USB camera, you will be able to use the UWP Camera APIs to consume camera information. With that release, we will also have camera samples. I do not have an ETA for that release.

closed 2015-06-21 by ooeygui

elmerm, 2015-06-22:
That's a bummer. It's a shame that a native Raspberry Pi component like the camera is not going to be supported.

ooeygui, 2015-06-22:
Hi @elmerm,
I will follow up and see what can be done.

elmerm, 2015-06-22:
Thanks @ooeygui.

mms-, 2015-09-17:
CPU usage is very high when let say a LifeCam Studio is connected to a Raspberry Pi. At 1080p not much room is left for other computation. This is because the CPU is not as powerful as a normal PC, however is there room for optimization?

Perhaps the driver model on Windows can be optimized to support a very low fps, i.e. 1 or 2 fps but full resolution? For some applications that will suffice. Otherwise it's a big waste running the camera at higher fps, having it consume CPU, and power, only to take 1 frame a second from the stream.

Even better, would be if the camera APIs could do 30fps at 480p while having a simultaneous stream of 1 fps at 1080p. With the computation done on the GPU by the platform since the cameras can't do this. Though I suspect UVC is a bit primitive to do this efficiently?

bfjelds, 2015-09-18:
We are still working on implementing the graphics driver for the Raspberry Pi.  This results in many GPU operations being handled by software.  You are seeing the results of that first hand.  Be assured that we are well aware of this and working to fix it!

IronManMark20, 2016-01-29:
Any eta on the graphics driver being released? I am hoping to use the pi for a webcam, and currently, it just doesn't do.

jessekaplan, 2016-01-29:
This is a priority item for us, with work underway, but unfortunately we don't have a timeline to announce at this point. 

I normally hate suggesting spending more money as a workaround for a limitation of our product, especially a temporary one, but just in case this would be useful for you I wanted to let you know. We recently released support for the Dragonboard 410c with GPU acceleration (there were already windows drivers for that gpu). It does cost a bit more than the RPI2, but it has on-board wifi and bluetooth and so, if those features are important to you, the cost evens out a bit. 

Thanks for your continued interest, and please let us know if you have any more questions. 

--Jesse

IronManMark20, 2016-01-29:
Am I correct in that you are waiting on https://github.com/Microsoft/graphics-driver-samples for gpu acceleration? As I understand it, the issue with the video capture sample is that it needs hardware acceleration. Is this correct? Also, thanks for the suggestion, but the Pi is essentially a requirement.

jessekaplan, 2016-01-29:
Our team is actually working on the driver directly, we're not waiting on another project. You're understanding is correct though, the issue is that this scenario requires hardware acceleration.

--Jesse

artus9033, 2017-05-30:
So one year passed, is there any progress?