ms-iot/samples #40: I2C Compass sample

reneherrero, 2015-08-10:
Hey,

First off, thanks for the samples!  Very helpful...

Time to give something back: an I2C Compass sample that uses the Honeywell HMC5883L magnetometer IC.  I´m more of a TFS guy and I'm not sure how to submit the code over to you guys, so here´s the link from a repo I created for myself: https://github.com/reneherrero/Windows10-IoT-Samples/tree/master/I2CCompass

I put it under an MIT license, let me know if that doesn´t work...

Cheers!

René

bfjelds, 2015-08-10:
Thank you!  That is awesome!!!

We can work at pulling your project into our samples if you'd like, or if you want to better understand how to do it yourself (the benefit there is that you would get the credit for your contribution!!):  http://ms-iot.github.io/content/en-US/Contribute.htm

Let me know and I will either start migrating your project to our repo or start watching for your pull request...

(Another idea would be to put your project on hackster.io ... something like https://www.hackster.io/4360/i2c-accelerometer)

reneherrero, 2015-08-10:
Thank for the link on hot to contribute, I'll figure that out tonight.  I think keeping things synced up will be easier that way.

Note that the IC has many more capabilities that I haven't yet explored (self-test, adjusting the gain to increase accuracy, ...).  Also, the algorithm that converts the 16-bit 2’s compliment hex values returned by the IC to X, Y and Z values (which in turn are converted to a bearing) is off.

Also, here's where I got the PCB board from ($6 -> dirty cheap): http://www.amazon.com/gp/product/B00QAG7SH6
The board layout is a little odd as some of the markings are on the underside. You’ll find yourself flipping it around quite a few times to make sure it’s connected appropriately.  The nice thing is that it also exposes the DRDY pin (Data Ready, Interrupt Pin) allowing you to get the maximum output rate by monitoring the hardware interrupt pin (next in my TODO list).

I anyone knows where I can find tilt compensating algorithms, please let me know!

bfjelds, 2015-08-10:
I haven't done any tilt compensating, but I found this on stackoverflow.com:  http://stackoverflow.com/questions/24639529/tilted-magnetometer-output-pitch-and-roll-tilt-compensation-i-am-lost

Hope that helps!

closed 2015-08-12 by bfjelds