ms-iot/samples #44: md files

braegelno5, 2015-08-12:
Is it possible to have in each project a readme.md with a short project descriptions and dependencies?
:)Torsten

reneherrero, 2015-08-12:
Now you’re just being picky!  :)

 

Will do…

bfjelds, 2015-08-12:
Our strategy has been to create in depth descriptions on our documentation site (look under the Samples tab):

http://ms-iot.github.io/content/en-US/win10/StartCoding.htm

Please take a look!

braegelno5, 2015-08-12:
Thanks i got it. If this link is not in the Main MD file it would be helpfull to place them there also, i think.

bfjelds, 2015-08-17:
That makes a lot of sense!  We'll look into getting the link into our MD files.

djaus2, 2016-10-27:
Why not keep that file with the project code in the MS-IOT/Samples<proj> repository? That way updates to the documentation can be part of the specific project updates. Could have an automatic copy mechanism to https://github.com/ms-iot/content/tree/develop/en-US/Docs.

PS I am looking at contributing a new project to Samples. How do I currently get my <proj>.md file up into Docs?

zhuridartem, 2017-03-28:
Please see https://github.com/ms-iot/content/blob/develop/README.md on how to add new .md files for Docs.You would need to also update _data\_docs-index.json to add a link to new doc.

IoTGirl, 2017-06-20:
Closing Stale - Site has been overhauled and is scheduled for another soon.

closed 2017-06-20 by IoTGirl