ms-iot/samples #58: Display sleeping?

Dids, 2015-08-25:
I couldn't find anything regarding display sleeping, and as long as I'm running any of my apps the display doesn't sleep. I think I recall it sleeping with the IoTCoreDefaultApp, though.

Any chance of a code sample for manually setting the display to sleep/standby?

rachitb777, 2015-08-26:
You can use the powercfg tool to set display timeout property. You can run the following command from PS/SSH 
powercfg -x -monitor-timeout-ac 1
IoTCoreDefaultApp doesn't have any code for display standby.

Dids, 2015-08-26:
Thank you, that does indeed work! I had no idea that something like the powercfg tool was also available on IoT. :)

closed 2015-08-26 by Dids

Dids, 2015-08-26:
I don't suppose there's a way to hook up an event handler to detect standby/wakeup?

I'm essentially trying to trigger a pin on my Rasberry Pi, which will turn my display off (or on).

I'm currently using a countdown timer to do so, but I couldn't find a way to have a global tap/mouse handler for the entire app, so I'm manually triggering the pin when any buttons in my app are pressed.

rachitb777, 2015-08-27:
Thanks for the feedback. We are looking into adding APIs for power-management in our future releases.