ms-iot/samples #60: Serial sample -> No keyboard input on RPI2

piwi1263, 2015-08-27:
Don't know if more people are experiencing this behavior, but using the serial sample I succeeded in using an USB to UART Bridge Controller based upon CP2102 and send/receive some bytes forth and back using xbees but I couldn't type in text in the write box when running the app on an RPI2 although the track pad mouse does work. Keyboard used is a Logitech K400r.

Sumahitha, 2015-08-28:
Thank you for reporting this problem. We are currently investigating.

marcusrussell, 2015-11-09:
Hello pwi1263,

This behavior has been improved, and the resolving changes are available in current releases.  Keyboard input should be functional in the scenario you describe with the latest content. You can get the latest content from the following link: http://go.microsoft.com/fwlink/?LinkID=625104

IoTGirl, 2016-06-03:
Fixed

closed 2016-06-03 by IoTGirl