ms-iot/samples #61: AllJoyn Explorer doesn't complete introspection of a local producer.

ianlee74, 2015-08-29:
If I deploy a producer to a Pi, then I am able to inspect it and call its methods via AllJoyn Explorer.  However, if the same producer is running on the same local machine then Explorer sees the producer fine but when trying to inspect its interfaces it times out or hangs.  AllJoyn Explorer is a great tool but because of this, it has limited use for me in demos.  Please fix or post the source code so that I can attempt to myself.  Thanks!

zhuridartem, 2015-08-31:
Hello Ian,

For Windows Runtime apps, network communications using an IP loopback address for interprocess communication is restricted. This applies between two apps running on the same machine, but does not apply if running on different machines or within the same process. You can learn more about the loopback restrictions here: https://msdn.microsoft.com/en-us/library/windows/apps/Hh780593.aspx

In your setup, sounds like you have AllJoyn Explorer UWP app and your AllJoyn Producer UWP app running on the same machine. AllJoyn allows for discovery of the service, but network loopback restriction prevents establishing an AllJoyn session which is required for introspection and enumerating objects.

Note that for AllJoyn for Windows 10 IOT Core you CAN run multiple apps and they will be able to communicate on the same device via AllJoyn. For Windows 10 Desktop, however, you will need to enable a loopback exemption for both apps.

A loopback exemption must be enabled for both apps: AllJoyn Explorer and AllJoyn Producer.

There are two ways to enable network loopback exemption:
1. Using Visual Studio: When building and deploying your app, look for and enable "Allow Local Network Loopback" option in project settings. This will ensure when an app gets deployed it also gets loopback exception enabled.  
2. Using built-in CheckNetIsolation.exe tool: At any point you can enable loopback exemption of an installed/deployed app. For specific steps for AllJoyn Explorer see Troubleshooting section of the https://github.com/ms-iot/samples/blob/develop/AllJoyn/AllJoynExplorer/AllJoyn_Explorer_Setup_Guide_v1.0.pdf?raw=true
 
Please let us know if these steps worked for you! 
Artem.

ianlee74, 2015-09-02:
Artem,

Thank you for the explanation. I was able to get it working by enabling the loopback exemption for the AllJoyn Explorer application.  Unfortunately, this was not well documented in the manual and makes an assumption that someone already knows the package ID of that application.  I worked out a process that is easier to follow.  You may want to update the manual.

1. Start AllJoyn Explorer.
2. Open Task Manager and find AllJoyn Explorer in the Processes list.
3. Right-click on AllJoyn Explorer and select "Properties".
4. In the General tab, you will a name such as "**adff22a6-72bc-4137-be64-04899205e0a7**_1.0.1.5_x64_**_hk7qwdpjbkp1m**".
5. Open a Command Prompt (in Administrator mode).
6. Type "CheckNetIsolation.exe LoopbackExempt -a -n=adff22a6-72bc-4137-be64-04899205e0a7_hk7qwdpjbkp1m".
Note that the last parameter is the combination of the two highlighted portions from # 4.
7. Restart AllJoyn Explorer and it should be able to successfully introspect your producer's interfaces.

Do you have any plans to release the source code for this sample app?  I'm building something similar for my needs.  This would have been a great starting point.

Thanks!

Ian

closed 2015-09-02 by ianlee74