ms-iot/samples #63: DigitalSignageUAP - Does not work on smaller screens

jpoon, 2015-09-03:
App not coded for different screen sizes; gets cut off on 7" display.

bfjelds, 2015-09-03:
Thank you for raising this issue!  We will look into a fix for this.

ricl, 2015-09-24:
None of the IoT sample apps run well/at all on 800*480 screens (which is a common screen size for the Pi).

SlavyMihov, 2015-09-24:
This issue has been brought to the attention of the app contributors and will be addressed.

IoTGirl, 2016-06-03:
Feedback taken by IoT Team member - Closing

closed 2016-06-03 by IoTGirl

jpoon, 2016-06-03:
@IoTGirl what does that mean? Are the apps being updated to support running on smaller screens?

SlavyMihov, 2016-06-03:
Yes, the apps have already been updated to display properly on small screens.

jpoon, 2016-06-03:
Cool. 👍 

gitetiquette: link the issue with the PR.

IoTGirl, 2016-06-06:
@jpoon I would normally link to the fix, but in this case it looks like the fix has been in for quite some time so we do not have a direct change that we can point to.