ms-iot/samples #64: Zip files of source code in the sample repo

jpoon, 2015-09-03:
https://github.com/ms-iot/samples/blob/develop/AllJoyn/AllJoynZWaveDemo/ZWaveAdapter.zip
https://github.com/ms-iot/samples/blob/develop/AllJoyn/AlljoynMockAdapter/MockAdapter.zip

These are zips of source code. Please expand these zips and put the source codes straight into git?

TorstenStein, 2015-09-03:
The zip files are provided to contain only the code needed for the sample. You can find the complete open source tree here: https://git.allseenalliance.org/cgit/dsb.git

jpoon, 2015-09-03:
It's strange to have a zip file of source code within a repository as you it prevents browse/search/maintain history of the code through git and GitHub.

As a suggestion, as the dsb.git is a git repo, add these as a git submodule?

zhuridartem, 2015-09-04:
Hello, and thanks for the feedback. Doing the cross reference as git submodules is a good idea. We did not get around to this yet.

The zip files do come from another git repo that contains more than the samples mentioned. Each zip here is made to be self-contained. We will evaluate the git submodules option.

dotMorten, 2015-10-13:
This pattern is extremely annoying and prevents us from seeing the history here, bloats the git database and the download. Also the AllJoynExplorer app doesn't even have source code, but just the appx.
If these aren't samples, but copies from elsewhere, what are they doing here in the sample repo?

ooeygui, 2015-11-06:
@zhuridartem is working on a solution for this.

zhuridartem, 2015-11-17:
Fixed. Thank you for your patience!

closed 2016-03-11 by zhuridartem