ms-iot/samples #65: Windows Service for IoT ?

S3NS4Ti0N, 2015-09-03:
Is it possible to write a Service for the RPI 2 ? We want to Update automatically update our application from our repositorys.

bfjelds, 2015-09-04:
Servicing is a scenario that we are thinking a lot about currently.  It is definitely on our radar.  I don't have an answer for you right now, but will work at getting an answer for you ASAP.

JohnMasen, 2015-09-09:
Background App(Aka: headless app) may be an option. it has no UI and can start automatically with OS.
ps: you need use IoTStartup command to set your headless app run at startup.

mms-, 2015-09-15:
Common scenarios for a Windows Service or a Background App are 'is the app running and healthy if not restart' and 'update the app'. Both of these should be handled by the platform. 

That is Windows IoT could (does?) have an update mechanism, something like ClickOnce (but reliable) or APPX or whatever. It should also have a durability feature, so an ability to set an app as foreground app, meaning it is always running and visible (in the case of a headed app) and is actively being checked for to enforce this, not just on start-up. It should also have a basic health query API, if the app does not respond it is shutdown and restarted.

bfjelds, 2015-09-17:
Thank you for the feedback!  

When an app is specified as a startup app (it has been configured to be a foreground app that is always running and visible), the platform is ensuring that this app is always running.  You can see how to configure the Startup App in the 'Set HelloWorld as the Startup App' section of [here](http://ms-iot.github.io/content/en-US/win10/samples/HelloWorld.htm).

We are working to bring more system management APIs into our platform and appreciate the suggestions for health query, servicing (update), etc.

mms-, 2015-09-17:
Good to know that always running check is already in there. I haven't had time to play around with the start-up functionality yet.

One un-related issue that we ran in to is the high CPU usage on Raspberry Pi with webcams : https://github.com/ms-iot/samples/issues/30

bfjelds, 2015-09-18:
Regarding the CPU usage, we are still working on implementing the graphics driver for the Raspberry Pi. This results in many GPU operations being handled by software. You are seeing the results of that first hand. Be assured that we are well aware of this and working to fix it!

S3NS4Ti0N, 2015-09-26:
Hey guys,

thank you for your fast reply.

@JohnMasen : Background Apps, are not suitable for my needs, i have to access the filesystem, in order to update my application. As far as i know, you cannot access the filesystem as from a normal UWP application. If i missed something, please point me to the right direction :)

@mms- : Good mention with the ClickOnce. That would be very handy to use. Another possible way would be, to publish the application to the normal store, so others can install them on their devices as well, from like Webb ?! maybe ?. I think it is just easy as a pie, for the @msftgits guys to add a new category or something like that. I think that would be a plus for the new eco system, microsoft is trying to implement with iot.

Cheers Guys !

S3NS4Ti0N, 2015-10-19:
Hey guys. Any updates here ?

bfjelds, 2015-10-22:
We created a simple C++ service based on the C++ Console Application template that can be used to create a service for IoT Core:

https://github.com/ms-iot/samples/pull/119

Hope this helps!!

mms-, 2015-11-02:
Will try the C++ service soon. What is the recommended way to run C# code within that? Or can the service also be C# based without loss of functionality?

bfjelds, 2015-11-10:
Only C++ is supported for console applications and services currently.

closed 2015-11-10 by bfjelds

S3NS4Ti0N, 2015-11-26:
When will VB.Net or C# will be supported ?

mms-, 2015-11-26:
Is it managed C++ that can call managed c# assemblies?

bfjelds, 2015-11-30:
Outside of WinRT, we don't have plans to support managed code on IoT Core.

For the service sample, it is not managed C++.

mms-, 2015-11-30:
That makes sense, I assume then that WinRT will grow in capability and WinRT based services or equivalent will arrive at some point. Hopefully not years..