ms-iot/samples #67: C# DSB Adapter will create a device with 0 interfaces when you add 2 AdapterValue to AdapterDevice.Attributes property

JohnMasen, 2015-09-09:
I modified the DSB C# template (Adapter.cs) as following:
```csharp
        public uint Initialize()
        {
            devices.Add(createTestDevice());
            return ERROR_SUCCESS;
        }
        private AdapterDevice createTestDevice()
        {
            string serial = new Random().Next().ToString();
            AdapterDevice result = new AdapterDevice("DeviceName", "Vendor", "Model", "version", serial, "description");
            AdapterProperty p1 = new AdapterProperty("P1", string.Empty);
            AdapterValue v1 = new AdapterValue("v1", 0);
            p1.Attributes.Add(v1);
            AdapterValue v2 = new AdapterValue("v2", 0);
            p1.Attributes.Add(v2);//this line will cause the DSB creates the device with 0 interfaces

            result.Properties.Add(p1);
            return result;
        }
```
it seems adding more than one AdapterValue will cause the device has 0 interfaces.

[image]
[image]

StepMan91, 2015-09-09:
Thank you !  very good screenshot

zhuridartem, 2015-09-10:
Hi Folks! I've reproduced this issue, thank you for reporting this. The screenshots are great!

What is happening is when you add a new AdapterValue to the property Attributes, the generated AllJoyn interface (which is based on the Property name and Attributes content) changes. But the interface name is the same. 

So, if you have some application like AllJoyn Explorer that has seen the other version of the interface with the same name, it gets confused. To fix, simply restart the consumer application (AllJoyn Explorer) and you should see the new version of the interface with updated properties.

JohnMasen, 2015-09-11:
thanks guys, this answered my question. I can continue work with my MQTT_DSB now :)

ooeygui, 2015-11-06:
Closing the issue here while we fix the issue in Alljoyn Explorer.
Thanks for reporting!

closed 2015-11-06 by ooeygui