ms-iot/samples #72: Working with MCP3008

sebagomez, 2015-09-16:
In your `convertToInt` function of the [TempSensor sample](../blob/develop/TempSensor/CS/MainPage.xaml.cs) file, you have a comment that reads the following:

    /*Uncomment if you are using mcp3208/3008 which is 12 bits output */

and actually, I read the MCP3008 is 10 bits... totally noob question here, but what should I use with the MCP3008? Is it a bug in the documentation or I just misunderstood what I've read so far?

Thanks

TychosNose, 2015-09-17:
Thank you for pointing out this issue. You are correct in that the MCP3008 is 10-bits and yes it is a bug but with some fixes you should be able to get the MCP3008 working with this code.

First, the comment
     ` /*Uncomment if you are using mcp3208/3008 which is 12 bits output */ `
 should be
     ` /*Uncomment if you are using mcp3204/3208 which is 12 bits output */ `

Second, there doesn’t appear to be code for the MCP3008 in this sample. However, with some modifications this code can work with the MCP3008. Here is what you need to do.

For reference, [here is the MCP3008 datasheet] (http://www.microchip.com/mymicrochip/filehandler.aspx?ddocname=en023833). Take a look at Figure 6-1 in the datasheet for reference. I’ll assume you are using channel 0 (Pin1) on the MCP3008.

1) Replace this code:

    /*Uncomment if you are using mcp3002*/ 
    int result = data[0] & 0x03; 
    result <<= 8; 
    result += data[1];

With this code:

    /*Uncomment if you are using mcp3008*/ 
    int result = data[1] & 0x03; //note we are using the second byte here instead of the first
    result <<= 8; 
    result += data[2]; //note we are using the third byte here instead of the second

2) Find the following code which should be around line 100 in the same file

    /*Uncomment if you are using mcp3002*/ 
    byte[] readBuffer = new byte[3]; /*this is defined to hold the output data*/ 
    byte[] writeBuffer = new byte[3] { 0x68, 0x00, 0x00 };//01101000 00; 

And replace with this code:

    /*Uncomment if you are using mcp3008*/ 
    byte[] readBuffer = new byte[3]; /*this is defined to hold the output data*/ 
    byte[] writeBuffer = new byte[3] { 0x01, 0x80, 0x00 }; //00000001 10000000 00000000

Good luck! Let us know if you have any more issues.

sebagomez, 2015-09-17:
Cool thanks. Will let you know how it went

ooeygui, 2015-11-06:
This has been captured in the sample. Thanks for reporting!

closed 2015-11-06 by ooeygui