ms-iot/samples #73: Memory leak in IoTCoreDefaultApp

ricl, 2015-09-23:
Running IoTCoreDefaultApp on both the Pi and the Desktop creates a memory leak when navigating to/from the Tutorials page.

Reproduction:

Run IoTCoreDefaultApp and observe the memory used (via Visual Studio on the Desktop, http://192.168.1.19:8080/SystemPerformance.htm on the Pi).

Navigating between Tutorials and Device Info (and around inside Tutorials) causes memory to be allocated and never released. This is even true after GC.

bfjelds, 2015-09-23:
Thank you for finding this and bringing it to our attention!  We'll work on fixing this ASAP.

ricl, 2015-09-24:
It would appear that both

            Frame rootFrame = Window.Current.Content as Frame;
            rootFrame.Navigate(screenType, parameter);

and

            Frame rootFrame = Window.Current.Content as Frame;
            rootFrame.GoBack();

ALWAYS call the constructor on the Page referenced. This is regardless as to whether the Page is in the BackStack or not.

ricl, 2015-10-01:
Solved:

Add

...
            this.InitializeComponent();
            this.NavigationCacheMode = NavigationCacheMode.Enabled;
...

to all the pages to allow for correct BackStack implementation on IoT Core.

Constriuctors are then not called on .Navigate(...) or .GoBack()

bfjelds, 2015-10-01:
Thank you for sorting this out!  We will work to update our samples accordingly.

bfjelds, 2015-10-13:
This has been addressed with https://github.com/ms-iot/samples/pull/98

closed 2015-10-13 by bfjelds