ms-iot/samples #78: IoTCoreDefaultApp Language Setting does not change input locale

PatGet, 2015-09-25:
IoTCoreDefaultApp Language Setting does not change input locale. For example the second part of the OOB Expirience is WIFI Network Password. All the UI there is presented in German language to me but I have to type in the Password as if i had an english Keyboard.

grahamehorner, 2016-03-04:
@PatGet have you performed a reboot after changing the Language Settings; I had something similar with regards to date/time region; and it's because of the system/thread culture change isn't detected by the app.

A restart stored it for me

grahamehorner, 2016-03-04:
https://github.com/ms-iot/samples/issues/147

bfjelds, 2016-03-07:
This locale change issue is a problem with the OS currently that we are working to fix.

rachitb777, 2016-07-14:
We have added Keyboard layout setting in the IoTCoreDefaultApp. Skip OOBE and then goto settings to change the keyboard layout.

closed 2016-07-14 by rachitb777