ms-iot/samples #82: 433 Mhz sample

vtni, 2015-09-28:
Hi guys,
a sample for receiving and sending 433 Mhz codes would be nice!

marcusrussell, 2015-11-10:
Hello VTNI,

Thanks for your feedback.  We are considering developing a 433 MHz code sample and will update this post with our plans and progress.

Thanks

vtni, 2016-02-08:
Hi,
are there any news about this?

EdiWang, 2016-06-06:
I have got a 315 MHz working, it's very straight forward, when your remote send signal to the receiver, the corresponding pin will output GPIO.High, so in your code, just check for the particular GPIO Pin input, there's really nothing special here. I think it's the same thing with 433 MHz also.

I have write a blog post about this http://edi.wang/post/2016/5/21/windows-10-iot-remote-door-control but it's in Chinese. Hope it will help.