ms-iot/samples #95: half font on ssd1306 128x32

sely2k, 2015-10-06:
hi
i've tried to use the library "spi display" with my oled ssd1306 128x32 (not 128x64) and not work
i've modified this line 

        private const UInt32 SCREEN_WIDTH_PX = 128;                         /* Number of horizontal pixels on the display */
        private const UInt32 SCREEN_HEIGHT_PX = 32;                         /* Number of vertical pixels on the display   */
        private const UInt32 SCREEN_HEIGHT_PAGES = SCREEN_HEIGHT_PX / 4;    /* The vertical pixels on this display are arranged into 'pages' of 8 pixels each */

but this library don't work with my display
(see the photo) http://1drv.ms/1OjAP6p
all the font are "half" 

can you help me?

dankuo, 2015-10-07:
Hello,

Try using these values instead:
```cs
private const UInt32 SCREEN_WIDTH_PX = 128;                         /* Number of horizontal pixels on the display */
private const UInt32 SCREEN_HEIGHT_PX = 32;                         /* Number of vertical pixels on the display   */
private const UInt32 SCREEN_HEIGHT_PAGES = SCREEN_HEIGHT_PX / 8;    /* The vertical pixels on this display are arranged into 'pages' of 8 pixels each */
```
You were correct in changing SCREEN_HEIGHT_PX to 32. However SCREEN_HEIGHT_PAGES should remain at its original value. This calculates the number of 8-pixel tall pages on the screen, so we continue to divide by 8. Also since your display is only 32 pixels tall, you only get lines 0 and 1 for text. You'll need to comment out the code where the sample tries to write out lines 2 and 3.

```cs
/* Update the SPI display to mirror the textbox contents */
private void DisplayTextBoxContents()
{
	try
	{
		ClearDisplayBuf();  /* Blank the display buffer             */
		WriteLineDisplayBuf(Display_TextBoxLine0.Text, 0, 0);
		WriteLineDisplayBuf(Display_TextBoxLine1.Text, 0, 1);
		//WriteLineDisplayBuf(Display_TextBoxLine2.Text, 0, 2); // Comment this out
		//WriteLineDisplayBuf(Display_TextBoxLine3.Text, 0, 3); // Comment this out
		DisplayUpdate();    /* Write our changes out to the display */
	}
```
With these changes, you'll only be able to use the first two textboxes in the UI to input text. Once you have the display working, you can remove those two lines from the UI.

Regards,
Daniel

sely2k, 2015-10-07:
unfortunatly still not work

dankuo, 2015-10-08:
What are you seeing on the screen now?

sely2k, 2015-10-08:
this nigth i've solve a lot of problem :) asap i'll fork the project and i'll make a pull request!

closed 2015-10-12 by sely2k