ms-iot/samples #105: GPIO OneWire timeout with DHT22

jmservera, 2015-10-09:
I have several DHT22 sensors and tried with different rpi2 in OS Version 10.010531.0 and also tried to change the wiring with a pullup and with different GPIOs too, but the code doesn't read the sensor, I always get timeout, seems that it doesn't have precission enough for the ms delays.

jessekaplan, 2015-10-09:
The DHT22 sensor has timing requirements that require a sequence of GPIO operations that are just on the border of what can be accomplished with the current product. It's not so much that there isn't enough precision, but rather that the operations take too long and so you miss the window. It's actually *right* on the border of what that version of IoT Core supports. You can look at the code in this sample for a sample that often succeeds (~1/3 tries): https://github.com/ms-iot/samples/tree/develop/GpioOneWire

We are working on options to improve this performance, but nothing we can announce quite yet. 

Thanks,
Jesse

jessekaplan, 2015-11-09:
Now we do have something we can announce :-)

The latest "Insider Preview" has significant performance improvements to the Gpio operations that should make it much easier to use the DHT22. We see ~3x improvements in speed over the 10240 build and the Fall Release will have another 50% on top of that. 

Additionally, this preview also includes an optional driver that, when enabled, increases performance more than 100x above the 10240 build. You can find information on this here: http://ms-iot.github.io/content/en-US/win10/LightningProviders.htm

Please let us know if that solves your issues. 

Thanks,
Jesse

closed 2015-11-25 by marcusrussell