ms-iot/samples #112: AllJoyn samples broken

dotMorten, 2015-10-13:
ZWaveAdapter is not compiling now:
```
2>c1xx : fatal error C1083: Cannot open source file: '..\..\windows\winversion.cpp': No such file or directory
```

Latest AllJoynExplorer explorer app won't deploy when executing powershell script:
```
Windows cannot install package AJX.adff22a6-72bc-4137-be64-04899205e0a7 because this package depends on another package that couldn't be found. This package requires minimum version 1.0.23302.0 of framework Microsoft.NET.CoreRuntime.1.0 published by CN=Microsoft Corporation, O=Microsoft Corporation, L=Redmond, S=Washington, C=US to install; the frameworks with name Microsoft.NET.CoreRuntime.1.0 available to the user are Microsoft.NET.CoreRuntime.1.0_1.0.23302.0_x86__8wekyb3d8bbwe. Provide the correct framew
Windows cannot install package AJX.adff22a6-72bc-4137-be64-04899205e0a7 because this package depends on another package that couldn't be found. This package requires minimum version 1.0.23302.0 of framework Microsoft.NET.CoreRuntime.1.0 published by CN=Microsoft Corporation, O=Microsoft Corporation, L=Redmond, S=Washington, C=US to install; the frameworks with name Microsoft.NET.CoreRuntime.1.0 available to the user are Microsoft.NET.CoreRuntime.1.0_1.0.23302.0_x86__8wekyb3d8bbwe. Provide the correct framework along with this package.
```

Now if we only had source code instead of zips for the first one, we could see the history and fix it, and if we had actually source code for the second one, we could just F5-deploy from Visual Studio, but alas.

dotMorten, 2015-10-13:
@zhuridartem

zhuridartem, 2015-10-14:
Hi Morten. Thank you for trying out these new bits, and we appreciate the feedback. 

Our initial thinking was to make the samples fully self-contained, thus the .zip files. We have heard of making the AllJoyn samples as source and were pointed to use the git submodule to reference other repositories. This makes sense, we have not gotten to refactoring these sames to work with that. 

Sorry to hear the latest ZWaveAdapter is not working for you. I've tried to reproduce the not compiling issue, and it built without this error. Sounds like the project that is failing to build is the .\ZWaveAdapter\open-zwave\cpp\build\winRT\vs2015\OpenZWave.vcxproj. The location of this project was moved from the previous version, so, could you please confirm that the older version of the project is not being used. 

I am assuming this is reproducing on the clean copy of the latest ZWaveAdapter. Could you share more details of the build output, as well as the version and flavor of Visual Studio that you are using? 
The version of VS I used for repro is 14.0.23107.0, with Windows 10 SDK (Visual Studio Tools for Universal Windows Apps) 14.0.23121.00.

We will investigate the AllJoynExplorer deployment issue in more detail. 

Thanks again,
Artem.

zhuridartem, 2015-10-29:
Update: we have reproduced the ZWaveAdapter not building. The issue is with the OpenZWave project's pre-build step that generates the winversion.cpp. To fix, in OpenZWave.vcxproj, please add missing quotes around $(ProjectDir)\..\..\windows\GIT-VS-VERSION-GEN.bat.

E.g.
    <PreBuildEvent>
      <Command>del ..\..\windows\winversion.cpp
CALL "$(ProjectDir)\..\..\windows\GIT-VS-VERSION-GEN.bat" "$(ProjectDir)\" "..\..\windows\winversion.cpp"
Xcopy /E /I /Y ..\..\..\..\config config\
exit 0</Command>
    </PreBuildEvent>

This will be fixed in next update of ZWaveAdapter.

Car-Sun-Key, 2015-10-31:
Just to be clear for others:

Within the OpenZWave.vcxproj file there are several instances of this line:

** CALL $(ProjectDir)\..\..\windows\GIT-VS-VERSION-GEN.bat "$(ProjectDir)\" "..\..\windows\winversion.cpp" **

which will need to be changed to this (add quotes around first path):

CALL "$(ProjectDir)\..\..\windows\GIT-VS-VERSION-GEN.bat" "$(ProjectDir)\" "..\..\windows\winversion.cpp"

zhuridartem, 2015-11-17:
This is now fixed. Please let us know if you have any issues!

MaheshGaware, 2015-12-03:
I am also facing this issue while I try to build and deploy UWP app with debug build. In release build, it works fine.

zhuridartem, 2016-01-19:
This will be updated with https://github.com/ms-iot/samples/pull/201

closed 2016-03-11 by zhuridartem