ms-iot/samples #114: Aljoyn samples with Gen5

Car-Sun-Key, 2015-10-14:
Anyone using Gen5 Z-wave hardware successfully?

zhuridartem, 2015-10-16:
We have not tried yet. Are you having problems with it?

Car-Sun-Key, 2015-11-01:
Hardware issue: When running AllJoyn Explorer I only see the zwave bridge (minJon - zwave bridge)  and the Pi (minJon). I currently have 2x Z-wave appliance modules and 1x Z-wave multisensor plugged in and paired - none displayed in AllJoyn explorer

Note: The tutorial shows different project names than what is provided in the sample. I'm assuming tutorial:ZwaveBackgroundService = samples:HeadlessAdapter app.

Question: What is the Z-wave headless app that is included in the stock image (and running by default)? Should this be stopped before deploying the headless z-wave app in the tutorial instructions?

Enviro: Win 10.0.10240, VS2015, fresh unmodified samples downloaded 10/31 (aside from the winver.cpp fix)

Thanks!

zhuridartem, 2015-11-17:
Thanks for the feedback. The ZWaveAdapterApp that is part of the stock image is indeed https://github.com/ms-iot/samples/tree/develop/AllJoyn/Samples/ZWaveAdapter 's HeadlessAdapterApp. 
You can disable the app from running by default by running in SSH or in WEBB's Process view the following command: "iotstartup.exe remove headless ZWave" and restarting your device. Now it should not start on boot.

dotMorten, 2016-01-13:
The Gen5 adapter uses a different serial device driver, which I'm pretty sure doesn't work with IoT Core.

AndreyShpilevoy, 2016-02-21:
It will be supported in the near future?
As I know it's very common device, it very pity, that it not supported yet.

dotMorten, 2016-02-21:
@AndreyShpilevoy Make sure you vote (or suggest if it's not already there) on connect: https://connect.microsoft.com/windowsembeddedIoT/Feedback

AndreyShpilevoy, 2016-02-21:
@dotMorten thank you for the link. I just created the task on this issue.
https://connect.microsoft.com/windowsembeddedIoT/Feedback/Details/2388618

IoTGirl, 2016-03-14:
Please use "https://insider.windows.com/" for feedback on IoT Pre-release images.

dotMorten, 2016-03-14:
@iotgirl the location that @AndreyShpilevoy posted, is the place the IoT team keeps telling us to use

IoTGirl, 2016-03-15:
@dotMorten That poster is incorrect. Windows IoT Core no longer uses Connect for distribution and has shifted to Windows Insiders to align with the other versions of Windows 10.

zhuridartem, 2016-05-17:
We will not be adding the UZB in-box driver to support the Gen5 in the near future. 
From http://aeotec.com/z-wave-usb-stick/1358-z-wave-drivers.html it does not look like they have a published ARM driver. 

I don't have the Gen5 hardware on hand.
 Perhaps you can manually try getting it to work by following these steps:
1. Get the driver from http://aeotec.com/z-wave-usb-stick/1358-z-wave-drivers
2. Copy the uzb.inf and uzb.cat to IOT Core device (x86 or amd64)
3. Follow https://developer.microsoft.com/en-us/windows/iot/win10/samples/driverlab3 to install the driver on IOT Core device.

Don't know if the OpenZWave stack that is in the sample also needs an update to support Gen5 stick.

IoTGirl, 2016-06-08:
Closed based on @zhuridartem's response

closed 2016-06-08 by IoTGirl

Maarten88, 2016-06-15:
We tried to get the Sample to work with our Gen 5 Z-Stick on Pi 2 and could not get it to recognize the device. Tried installing manually, copying .inf files as per the suggestion above, but could not make it work.

I think this is a problem, because the Series 2 hardware is getting hard to find, while the Gen 5 is everywhere. It's hard to follow the sample when you can't buy the hardware to make it work.

Any suggestions on how to get a working UZB driver for IoT on ARM?

Maarten88, 2016-06-15:
Tnx @AndreyShpilevoy, that was exactly what I tried, and indeed it reports "Driver is running". So maybe USB does work. Yet the sample still does not report any devices/sensors... I'm stepping through it with the debugger (my c++ is very rusty) and see no obvious errors, but it does not find any sensors (in IoT Explorer for Alljoyn)... Is there anything stupid I might be missing?

TheFlyingDutchMark, 2016-06-19:
I also installed the driver for the Z-stick gen 5 and it reports "Driver is running".
I followed these steps to setup the device: https://developer.microsoft.com/en-us/windows/iot/win10/samples/zwavetutorial
It says "Launch IoT Explorer for AllJoyn app. The app should find the following three devices:"

I stil don't see any device except the "ZWave Bridge"

dotMorten, 2016-06-20:
@TheFlyingDutchMark Is your z-stick correctly paired with your z-wave devices?

TheFlyingDutchMark, 2016-06-20:
@dotMorten 
Thank you for the response
When I plug in the z-stick in my pc I can see that the stick is paired with my smoke alarm. I'm using the tool from aeontec to diagnose the z-stick on my normal PC.
But when connected to my raspberry it doesn't show my stick at all in IoT Explorer for AllJoyn app

TheFlyingDutchMark, 2016-06-20:
@dotMorten 
Maybe you're right. I loaded the files and the app is running. I'm going to figure out this week if it finds something. I'll post an update.

TheFlyingDutchMark, 2016-06-21:
@dotMorten 
I ran the headed app from the source code and put breaks around line 646. To see what was going on I put a watch on "this pointer" in L645. I see the "ZWaveBridge" in there, but that's it. It doesn't find the Z-stick Gen5. So I think the result is the same as before with the raspberry and the IOT explorer.

I'm really wondering if there is somebody who has the gen5 working. I can't find anybody confirming it.

TheFlyingDutchMark, 2016-06-21:
I found something. When I add a watch for the string path, I see the value "VID_10C4&PID_EA60".
When I looked in windows device manager I found that the Z-Stick has as path "VID_0658&ID_0200".
So I manipulated the code by adding "path = "VID_0658&PID_0200";" to line 641. When I open the IoT explorer, I saw the ZWaveBridge is gone and the Z-Stick Gen5 is visible! But no paired device.

TheFlyingDutchMark, 2016-06-27:
I think I found a solution for the Z-Stick Gen5.
First I did what AndreyShpilevoy wrote:
"@Maarten88 check this article: http://jellebens.blogspot.com/2016_02_01_archive.html?m=1"

Additionally I edited the options.xml files in the project folder.
I changed the line "<Option name="ControllerPath" value="VID_10C4&PID_EA60" />" to "<Option name="ControllerPath" value="VID_0658&amp;PID_0200" />"

Now when I open IoT Explorer I can see the adapter!

TimmiT93, 2016-07-21:
I doubt if changing the ControllerPath is the right way to go. This is how I made it work.

The AllJoyn Z-Wave Bridge sample still didn't work with the Z-Stick Gen 5 after installing the drivers described in Jelle Bens his blogpost. What I did is replacing the 'config' folder in the OpenZWave project by this one: https://github.com/OpenZWave/open-zwave/tree/master/config. In this folder you can find all (newer) config files for Z-Wave devices, including the one for the Gen 5 stick. Of course you can also just copy the ones you need right now. In my case that were the zstickgen5.xaml and multisensor6.xaml files, which can both be found in the aeotec folder. Good luck! :)