ms-iot/samples #117: Add screensaver to DefaultApp

dotMorten, 2015-10-17:
When the default app runs for a long time, it has the risk of burning itself into the LCDs we connect to the devices.
It would be good to have a screensaver for the app to prevent this - I already have a sample done (and will be posting on hackster.io soon), so I would be happy to contribute this to the app, but before doing the work want to know if this is something that you agree on would be a good idea?

There is one down side of course: If you don't have a mouse, keyboard or functional touchscreen, once the screen saver has started, it's not possible to disable it again. However it might be possible to detect if any of these devices are available, and only if so, enable the screensaver.

stephenhuang, 2016-01-11:
Hi dotMorten,
I'm part of the Microsoft IoT Team. 
This is a great idea and we'd love to see your sample. Please submit a pull request and we'll take care of the rest.

dotMorten, 2016-01-11:
@stephenhuang Yes I can provide you with this, however, can you help guide me to solve the issue on the last paragraph?

stephenhuang, 2016-01-11:
@dotMorten My suggestion would be to have the screensaver disabled by default and to add an option to enable/disable it under settings (the gear in the top right hand corner).
In the coming months, we can take your work and improve it so it can be set remotely from a computer. This will solve your problem :)

dotMorten, 2016-01-12:
Got it! What do you want on the screensaver? Do you have a PNG or JPEG I can use? Perhaps a Windows IoT Core logo?

dotMorten, 2016-01-13:
@stephenhuang Alright! Check out the PR and let me know what you think.

stephenhuang, 2016-01-13:
Will do, thanks for helping out. We can always change the jpeg on our end (We'll need to comply with branding guidelines so it's best if we change it anyway).

Will take a look shortly

dotMorten, 2016-02-14:
@stephenhuang ping ;-)

stephenhuang, 2016-02-22:
Looks good. We'll pull this shortly

IoTGirl, 2016-06-15:
Confirmed - Closed

closed 2016-06-15 by IoTGirl