ms-iot/samples #128: Deployment of Win32 executables to c:\deploy

ricl, 2015-10-25:
It may be better to change this to something like

c:\PROGRAMS\MyAppsDir\MyProject 

for future expansion.

(The PROGRAMS directory is because that's where they always were!).

bfjelds, 2015-10-26:
Thanks for the feedback!  To be honest, when we put the Console application sample together, I don't think we gave a tremendous amount of thought as to the destination directory (especially given that it is completely configurable).  We will look into what the most appropriate location is for user scenarios (it might be more practical to aim at windows\system32?) and where we can access (given that MSVSMON is typically running as DefaultAccount, it may not be feasible to point at certain directories).

The reason that the Console Application requires more work than the UWP apps do is mainly an issue of prioritization.  The first priority was (and continues to be) getting UWP apps that can be developed and run across all Microsoft platforms.  As we progress, IoT will continue to work with Visual Studio to make our customers' priorities smoother and simpler!  Towards that goal, we will certainly think about how we can make developing the C++ Console Application a better experience.

ricl, 2015-10-26:
:-) The actual destination directory is of course up for grabs. The only
reasons for suggesting /PROGRAMS/... are

1. It's where they always were back in the day....
2. It's a path without spaces in its name (banish the risk of spaces now!)
3. It works currently in 10240 and 10556 with both Admin and User privileges
4. It provides a useful point at which to spawn the various sub directories
that individuals might like to accumulate from elsewhere (i.e. Manufacturer
named)

P.S. C applications are always the last....

ricl, 2015-10-26:
5. It means we don't need to step on the /Windows directory which might
muddy things

bfjelds, 2015-10-26:
That's great insight and I've added all of that to our bug tracking this!

ricl, 2015-10-26:
Or create a brand new directory from the root and move into a new world.
c:\IoTWin32 perhaps.

marcusrussell, 2015-11-25:
Assigning to Jesse, who has a task associated with this topic.

IoTGirl, 2016-12-05:
Hi Brian,
Has this issue been suitably captured so that it might be closed?
Sincere thanks,
IoTGirl

ooeygui, 2016-12-07:
We've captured this feedback internally and are working to improve the native experience. 
Thank you for the feedback.

closed 2016-12-07 by ooeygui