ms-iot/samples #132: DigitalSignageUAP will not deploy to Raspberry Pi

housten, 2015-10-27:
The döde compiles ok but I can't deploy it. I have set VS to ARM and Release and I was able to deploy the hello World APP.

This is my error: 

Severity	Code	Description	Project	File	Line
Error		Error : DEP0800 : The required framework "C:\Program Files (x86)\Microsoft SDKs\Windows Kits\10\ExtensionSDKs\Microsoft.Midi.GmDls\10.0.10240.0\.\Appx\Neutral\MidiVoices.appx" failed to install. 
error 0x800B010A: The root certificate and all intermediate certificates of the signature in the app package or bundle must be trusted.	DigitalSignageUAP		

Do I need to do something special with the included .pfx file? Or how can I get past this?

Thanks!

/Heidi Housten

ravindrank, 2015-12-10:
Additionally, in my case, immediately after getting the project from git, I could not build DigitalSignageUAP with an error indicating System.Runtime.dll missing. I had to remove the NETCoreUniversalWindow platform Nuget package and install it again, accept licence agreement during this process and then it builds.

SlavyMihov, 2015-12-10:
Thank you for your comments. I'll be taking a look at the issue. Can you describe me your setup? Which version of VS and SDK do you have installed?

ravindrank, 2015-12-11:
Hi SalvyMihov,

Sorry for my confusing post. I don't see any certificate errors etc like Housten sees.

I did see the System.Runtime.dll missing error. I remove the NETCoreUniversalWindows platform Nuget package and install it again, accept licence agreement during this process and then it built fine. I was able to successfully deploy as well.

I'm on VS2015 Update1.

SlavyMihov, 2015-12-15:
ravindrank, did you try building/deploying any of the other sample applications? Do you see the System.Runtime.dll missing error with any of them, or is the error specific to the DigitalSignage app?

ravindrank, 2015-12-16:
Hi,

Yes I see the same issue with Blinky as well..
We need to remove and readdd NETCoreUniversalWindows platform Nuget package and install it again, accept licence agreement ..

SlavyMihov, 2015-12-16:
In order to avoid uninstalling/reinstalling the NETCoreUniversalWindows platform Nuget package and have this automatically resolved when you first rebuild the solution, you can try enabling automatic package resolution by the NuGet Package Manager:
1. open Tools -> NuGet Package Manager -> Package Manager Settings
2. select NuGet Package Manager -> General
3. check "Allow NuGet to download missing packages" and "Automatically check for missing packages during build in Visual Studio"

IoTGirl, 2016-06-07:
Closing - No update for more than 1 month so I assume the issue is solved.

closed 2016-06-07 by IoTGirl