ms-iot/samples #133: Is it possible to view a pdf in the browser?

tomkuijsten, 2015-10-28:
I tried opening a pdf with the IoTBrowser sample, which won't work. Is it possible?

bfjelds, 2015-10-28:
The WebView does not support displaying PDF files.  Here is a summary of WebView info:  http://blogs.msdn.com/b/wsdevsol/archive/2014/04/03/ten-things-you-need-to-know-about-webview-_2d00_-an-update-for-windows-8.1.aspx.

There is a small mention of Windows.Data.Pdf which may work (I don't believe that anyone has tried that API yet on IoT Core).

tomkuijsten, 2015-10-29:
Got it working on the emulator, can't test yet on the pi, will do that soon but I have no reasons to believe it won't work.

```
private async void WebView_NavigationStarting(WebView sender, WebViewNavigationStartingEventArgs args)
{
    if (args.Uri.AbsolutePath.EndsWith(".pdf"))
    {
        try
        {
            // 1. Download the pdf
            var getPdfHttpClient = new HttpClient();
            var pdf = await getPdfHttpClient.GetByteArrayAsync(args.Uri.AbsoluteUri);

            // 2. Store it in a temp file
            StorageFolder folder = ApplicationData.Current.TemporaryFolder;
            StorageFile pdfFileOriginal = await folder.CreateFileAsync(Guid.NewGuid().ToString() + ".pdf");

            using (IRandomAccessStream pdfFileStream = await pdfFileOriginal.OpenAsync(FileAccessMode.ReadWrite))
            {
                await pdfFileStream.WriteAsync(pdf.AsBuffer());
                await pdfFileStream.FlushAsync();
            }

            // 3. Load pdf as PdfDocument
            var _pdfDoc = await PdfDocument.LoadFromFileAsync(pdfFileOriginal);

            // 4. Create png file
            StorageFile pngFileOriginal = await folder.CreateFileAsync(Guid.NewGuid().ToString() + ".png");

            // 5. As a test, only show first page
            using (IRandomAccessStream pngFileStream = await pngFileOriginal.OpenAsync(FileAccessMode.ReadWrite))
            {
                await _pdfDoc.GetPage(0).RenderToStreamAsync(pngFileStream);
                await pngFileStream.FlushAsync();
            }

            // 6. Set image as source for xaml image
            pdfImage.Source = new BitmapImage(new Uri(pngFileOriginal.Path, UriKind.Absolute));
        }
        catch (Exception ex)
        {
        }
    }
}
```

tomkuijsten, 2015-10-30:
Just tested it on a Raspberry Pi II and it works like a charm.

ooeygui, 2015-11-07:
Wow! Thanks for the report and closing the loop.

closed 2015-11-07 by ooeygui