ms-iot/samples #135: SerialPort USB Object reference not set

santmu30, 2015-10-29:
I am trying to find a solution to the Known Issue Object Reference not Set to an instance of an object when the usb is connected for the first time, I know that if I disconect the usb and reconnect, the problem is gone, but my enviroment is that the USB will be always connected and I need an unattended enviroment, so Please could you help me find a solution for it? i was trying to disable the USB port and enable again but i didnt find any function to it.

Thank You and Kind Regards

Santiago Muñoz

serusako, 2015-11-10:
Hi,
please try the following workaround - reboot your RPi2 platform and terminate the following process ZWaveHeadlessAdapterApp (shown in App Manager as running app)  before you plug in your USB serial adapter. You can also uninstall this app ZWaveHeadlessAdapterApp to make your changes persist across reboots.

zhuridartem, 2016-03-11:
ZWave App is no longer enabled by default. Closing issue.

closed 2016-03-11 by zhuridartem