ms-iot/samples #147: DefaultApp - Date Time

grahamehorner, 2015-11-06:
The DefaultApp is shows the Raspberry PI date time; but the Raspberry PI has no RTC; therefor the DefaultApp or Windows10 IoT core should look to set the date time using a NTP service at start-up

cmaloney-msft, 2015-11-09:
Hi Grahamehorner, Windows 10 IoT Core is configured to sync the time on boot as you suggest. 

Are you experiencing issues with the date/time?

grahamehorner, 2015-11-09:
If that is the case the Raspberry PI that is not receiving the time NTP on boot; have confirmed that this is the case when using WiFi as well as wired.

cmaloney-msft, 2015-11-09:
Can you share what build you are running and the output of the following command using a SSH or PowerShell console? 

w32tm /query /status

Thanks!

grahamehorner, 2015-11-09:
I'm running build 10.0.10556.0 and w32tm /query /status returned [minwinpc]: PS C:\Users\Administrator\Documents> w32tm /query /status
Leap Indicator: 3(last minute has 61 seconds)
Stratum: 0 (unspecified)
Precision: -6 (15.625ms per tick)
Root Delay: 0.1842967s
Root Dispersion: 8.0847694s
ReferenceId: 0x00000000 (unspecified)
Last Successful Sync Time: 11/9/2015 9:24:14 AM
Source: time.windows.com,0x9 
Poll Interval: 10 (1024s)

however the time is incorrect

grahamehorner, 2015-11-09:
I've also ran 

[minwinpc]: PS C:\Users\Administrator\Documents> tzutil /s 'GMT Standard Time'

[minwinpc]: PS C:\Users\Administrator\Documents> w32tm /query /status
Leap Indicator: 3(last minute has 61 seconds)
Stratum: 0 (unspecified)
Precision: -6 (15.625ms per tick)
Root Delay: 0.1842967s
Root Dispersion: 8.0847694s
ReferenceId: 0x00000000 (unspecified)
Last Successful Sync Time: 11/9/2015 5:24:14 PM
Source: time.windows.com,0x9 
Poll Interval: 10 (1024s)

several times the date/time is doesn't change?

cmaloney-msft, 2015-11-10:
Looks like your system is successfully contacting the time server. 

Can you tell if your issue a time zone issue and your clock is off by a hours exactly, or if your issue is a NTP/RTC issue and your clock is simply wrong? 

If it appears that it is a time zone issue can you confirm that your tzutil command took by running "w32tm /tz"

grahamehorner, 2015-11-10:
@cmaloney-msft local GMT time is 11:00am the Raspberry PI is showing 2:40am;  the w32tm /tz shows 

minwinpc]: PS C:\Users\Administrator\Documents> w32tm /tz
Time zone: Current:TIME_ZONE_ID_STANDARD Bias: 0min (UTC=LocalTime+Bias)
  [Standard Name:"GMT Standard Time" Bias:0min Date:(M:10 D:5 DoW:0)]
  [Daylight Name:"GMT Daylight Time" Bias:-60min Date:(M:3 D:5 DoW:0)]

grahamehorner, 2015-11-10:
OK, so after some checks the date and time only gets updated in the app after a reboot? you can change the timezone and datetime via remote powershell but it seems to have zero impact until the raspberry pi is reboot?

cmaloney-msft, 2015-11-10:
The time zone for the default app comes from the [CultureInfo.CurrentCulture] (https://msdn.microsoft.com/en-us/library/system.globalization.cultureinfo.currentculture%28v=vs.110%29.aspx) property which is set at thread start in this case. You can see this [here] (https://github.com/ms-iot/samples/blob/develop/IoTCoreDefaultApp/IoTCoreDefaultApp/Views/MainPage.xaml.cs)

However there the Clock in the default app should be accurate to the time-zone at boot. 

Can you confirm you are seeing the clock update (move/tick) a time different than that of the system time?

cmaloney-msft, 2015-11-13:
Hi Grahamehorner, did we get your questions answered?

grahamehorner, 2015-11-13:
Yes, thanks the Raspberry PI needs a reboot when changing timezone otherwise the DefaultApp will show the incorrect time, it would be good to allow the user to seect the timezone via the DefaultApp simlar to the language selection.

cmaloney-msft, 2015-11-13:
Thanks, I will make sure your feedback gets passed onto the team.

closed 2015-11-13 by cmaloney-msft