ms-iot/samples #149: Can't build BluetoothGatt sample

Mitwpgm, 2015-11-08:
[image]

Hi, Could someone know what might be wrong and how top resolve it?   I have updated my Visual Studio 2015 and all Windows10 SDK to the latest updates.  Both IDE and SDK are latest and up-to-date.  But I am still seeing the message as seen in screen-capture image when I load the project.  Can't build.  Please help.  Thanks!

wleev, 2015-11-09:
This example only runs on the latest [Insider Preview version of the Windows 10 SDK](https://dev.windows.com/en-us/downloads/windows-10-developer-preview), so you'll need that SDK plus an Insider Preview of the OS to run it. As far as I know Bluetooth pairing is not possible without the new Insider Preview, but you should be able to use the current SDK to receive/send data to already paired Bluetooth devices.

Mitwpgm, 2015-11-10:
Thanks wleev for that!  I have installed the latest Insider Preview version of the WIn10 SDK .  I can open and build project now.  However, I can't deploy to target device as the OS version is incompatible.  Where do you get the latest Insider Preview of the Win10 IoT OS?  I have downloaded the latest version of the Insider Preview for Raspberry Pi from  http://ms-iot.github.io/content/en-US/Downloads.htm .  However, that version of OS firmware is only 10.0.10556.0.  Where and how do you obtain version 10.0.10563.0 preview OS for Raspberry PI?  Thanks!

[image]

wleev, 2015-11-10:
There's no RPi OS yet that supports the API features in the BluetoothGatt project. You can't access Bluetooth from simulators either. You're only option is to get Insider Preview builds for your Windows 10 OS ( or possibly VM ). 

I've got the example working on my desktop, with some minor code changes. It appears DeviceInformationCustomPairing is already no longer supported on my current OS build ( 10586.0 ).

If you only want to prototype on the RPi, you'll have to do manual pairing until they release the new IoT Core image file.

kusumamurthy, 2015-11-10:
The changes in the Bluetooth GATT sample were pushed early. We are working on our next release and that release should work.

Mitwpgm, 2015-11-10:
Thanks again very much wleev for your helpful reply to my question!  I will try on desktop as suggested meantime while waiting for next release of the OS.

Kusumamurthy, thanks for the feedback on this sample project.  Could you please tell me when about is the next release going to be rolled out to public?  I wish to know the time frame I will have to wait for.  Thanks!

kusumamurthy, 2015-11-11:
Unfortunately, we cannot publicly announce any dates. We will update this thread when the sample can be run from the released bits.

ooeygui, 2015-12-07:
Please check with the latest release of Windows 10 IoT Core - it was released after this thread.

closed 2015-12-07 by ooeygui