ms-iot/samples #151: AllJoyn Explorer is broked when object registered with AJ_OBJ_FLAG_DESCRIBED

demon-xxi, 2015-11-09:
When alljoyn object is registered with AJ_OBJ_FLAG_DESCRIBED or with (AJ_OBJ_FLAG_DESCRIBED  | AJ_OBJ_FLAG_ANNOUNCED)  then AllJoyn Explorer shows 0 interfaces declared for the object. 

Even though AJ Explorer  does not yet support AllJoyn Events and Actions API, it should use basic Introspection interface which is still available for the object.

zhuridartem, 2015-11-17:
Could you tell us more about the setup that you are using? 
Specifically, are AllJoyn producers and AllJoyn Explorer running on the same machine? If so, please double check network isolation has been applied to both AllJoyn Explorer and producer application by following Troubleshooting section of the https://github.com/ms-iot/samples/releases/download/AllJoynExplorer_1.0.11/AllJoyn_Explorer_Setup_Guide_v1.0.pdf

demon-xxi, 2015-11-17:
@zhuridartem, I am using Router and thin client on Linux machine running 15.04 version of alljoyn. 
The explorer itself is running on windows in the same network.

IoTGirl, 2016-07-15:
Hi demon-xxi,
My apologies regarding the time it has taken to reply but the team is confident they know the source of the issue now and have added a work item to their Agile backlog.
My understanding is that this will involve changes to the AllJoyn Explorer to add missing interface descriptions but as the team is Agile, I can not give you a timeline for when this issue will be addressed.
Sincerely,
IoTGirl
Sincerely,
IoTGirl

closed 2016-07-15 by IoTGirl