ms-iot/samples #158: AllJoyn Explorer source code

koapeadu, 2015-11-16:
Hi,

I am trying to develop a UWP application using C# that will discover all AllJoyn producers on the network and create consumer objects for communicating with the devices(producers). It's going to be pretty much like the AllJoyn Explorer.

Is the source code for the AllJoyn Explorer available for developers?

If not can I get some insight or guides to writing such a program?

zhuridartem, 2015-11-17:
We are currently not planning to release AllJoynExplorer as source. 
However, AllJoynExplorer uses the following source to enumerate and interact with AllJoyn devices: https://github.com/ms-iot/samples/tree/develop/AllJoyn/Platform/DeviceProviders

dotMorten, 2015-12-09:
@zhuridartem Are you guys looking into providing this library as a nuget package for easy consumption? If not I'll be happy to do that, but figured I would ask the owners first. the DeviceProviders library is hugely useful for generic device consumers.

ianlee74, 2015-12-09:
+1 for making the AllJoynExplorer source available.  I've also started a customized version of something similar.  Given this is an app that basically everyone developing for AllJoyn on Windows uses, I think there would be great value in making it available as a reference project.

dotMorten, 2016-01-13:
I put up a nuget package of the DeviceProviders here: https://www.nuget.org/packages/dotMorten.AllJoyn.DeviceProviders I hope you're ok with this.
Still it would be great with an official package released by Microsoft, but I'm happy to maintain this package too.

zhuridartem, 2016-05-18:
Closing. 
Please use dotMorten's NuGet DeviceProviders or get the source from link above for generic AllJoyn enumeration.

closed 2016-05-18 by zhuridartem

koapeadu, 2016-06-02:
Thanks guys. You made my final year project a success. :+1: 
I also found [@dotMorten's tutorial on hackster.io](https://www.hackster.io/dotMorten/discovering-and-interacting-with-any-alljoyn-device-0dbd86) very useful.

dotMorten, 2016-06-02:
@koapeadu Glad you found it useful. I just published another library that builds on top of the device providers lib and exposes the common AllJoyn devices in an easy to use way: http://github.com/dotmorten/AllJoynClientLib