ms-iot/samples #163: AllJoyn for CoreCLR

grahamehorner, 2015-11-19:
Please consider creating an implementation of the AllJoyn provider and tools that is compatible with the CoreCLR

zhuridartem, 2016-01-19:
Could you clarify which tools are you referring to? Is this the AllJoyn Studio or the Device Providers?

grahamehorner, 2016-02-16:
the VSIX tools for AllJoyn

zhuridartem, 2016-07-15:
Latest implementation for DSB templates and AllJoyn Studio output for managed do support CoreCLR, which is required to run on IoT Core anyways. 

For DSB example, please take a look under https://github.com/ms-iot/samples/tree/develop/AllJoyn/Samples/ZigBeeAdapter/HeadlessAdapterApp. In project.json it calls out "Microsoft.NETCore.UniversalWindowsPlatform": "5.0.0" dependency. In HeadlessAdapterApp.csproj it specifies NETFX_CORE define constant, and references Universal CRT SDK. 

For AllJoyn Studio output example, please take a look at https://github.com/ms-iot/samples/tree/develop/IotOnboarding. The org.alljoyn.Onboarding project is generated using AllJoyn Studio, and is a C++/CX WinRT dll project. It can be directly consumed by a CoreCLR project and application, as seen in https://github.com/ms-iot/samples/blob/develop/IotOnboarding/IoTOnboardingService/IoTOnboardingService.csproj

Now, if you need to call to AllJoyn APIs directly from a managed project, that is a bit more involved. You have several options here - wrapping AllJoyn C API calls with C++/CX WinRT project, or wrapping AllJoyn C API calls with p/invoke definitions. The DSB BridgeRT.dll is doing the former.  For PInvoke see https://msdn.microsoft.com/en-us/library/aa288468(v=vs.71).aspx. 

Hope this helps!

closed 2016-07-15 by zhuridartem

grahamehorner, 2016-07-15:
I think that support for coreclr should be done in a platform agnostic way similar to the way the aspnetcore team use libuv for kestrel, .net should have a AllJoyn implementation abstraction layer, this allowing .net cross platform AllJoyn applications.