ms-iot/samples #171: DHT22/AM2302 Not Working with GPIOOneWire

HeyJude1997, 2015-12-02:
Currently I am using the DHT22/AM2302 which has a built in 5.1k resistor, using the program provided from https://ms-iot.github.io/content/en-US/win10/samples/GpioOneWire.htm and connected the sensors VCC to 3.3V(Pin1), GND to GND(Pin6), and DOUT to GPIO4(Pin7). Everything seems to be hooked up right but when running the program, no sensor is detected.  I've troubleshooted the sensor on a Linux platform and had no issues, others have reported that the DHT11 works fine with the program.  

I've been stuck on this issue for almost a month now with several threads on different forums open with no luck. If a clear answer is provided that would be much appreciated.

jordanrh1, 2015-12-03:
What build number are you running (type 'ver' at the SSH prompt)? The DHT11 sample only works on build 10556 and higher.

The sample is right on the edge performance-wise of being able to service the DHT11. If there is significant background activity (e.g. network, USB, disk, video activity) it could prevent the sample from working correctly.

HeyJude1997, 2015-12-03:
Currently running [Version 10.0.10240]. Here is the model number of the DHT22 as well as the kind I have bought. Model # (SN:150605AB6),  http://www.amazon.com/Temperature-Humidity-Measurement-Appliance-XYG-Study/dp/B0130I6MIC/ref=sr_1_8?ie=UTF8&qid=1449176521&sr=8-8&keywords=dht22

HeyJude1997, 2015-12-03:
That fixed it.

ooeygui, 2015-12-07:
Based on the last comment, this sounds like it works now.

closed 2015-12-07 by ooeygui