ms-iot/samples #176: Digital Signage Video not playing smoothly on 410c

bjnicholson, 2015-12-08:
video playback on the dragonboard 410c stutters.  It plays for 1-2seconds, then moves back several frames (1/4 to 1/2 of a second), and plays for 1-2 seconds before moving back 1/4 to 1/2 second again.  

This same issue happens when visiting a site in the browser of this sample with video embedded. 

In case it helps, it also happens outside of this sample app too (a simple app with just a video player) performs identically.  410 issue?

koapeadu, 2016-01-21:
I also have the same issue on the Raspberry Pi 2.

jwinarske, 2016-02-18:
I experienced the same on a 410c running the .63 build

IoTGirl, 2016-06-03:
Repro secured and issue logged : See MSDN: https://social.msdn.microsoft.com/Forums/en-US/63c8f470-7f63-4ea3-929f-c11bd61d1929/accelerated-video-on-dragonboard-410c?forum=WindowsIoT

closed 2016-06-03 by IoTGirl