ms-iot/samples #189: Http request is not read completely

tomkuijsten, 2015-12-22:
In the sample code there is a construction to read the IInputStream. I have a problem reading the full request, it sometimes stops before reading the body of the request. The issue can be found here:

https://github.com/tomkuijsten/restup/issues/20

And I found this SO issue (with a work-around):

http://stackoverflow.com/questions/32297416/c-sharp-http-server-using-streamsocketlistener-the-option-inputstreamoptions-p

Is there a solution for this problem?

tobiasstraub, 2015-12-28:
@tomkuijsten has found a solution for this problem. You can read about this [here](https://github.com/tomkuijsten/restup/issues/20#issuecomment-167320246) and show the [code here](https://github.com/tomkuijsten/restup/blob/19f989aaf194348f16de819e55ba696f9f6eac7f/src/WebServer/Http/HttpRequestStreamReader.cs).

closed 2016-07-14 by rachitb777

tomkuijsten, 2016-07-15:
Is it fixed?