ms-iot/samples #190: SerialDevice.LoadAsync  block await task

SonOfSam, 2015-12-28:
I having a little issue reading data from Nave32  Flight Controller, I'm able to send data to the card using the SerialDevice class but when the LoadAsync Method is call and there is no more data to read the app hung and does not move forward, there is not exception during the execution of the process. I look at the app state and there is a bunch a task block and I'm not sure why. See image below

here is my example code 

	namespace Apps.Receiver
	{
		/// <summary>
		/// An empty page that can be used on its own or navigated to within a Frame.
		/// </summary>
		public sealed partial class MainPage : Page
		{

			public MainPage()
			{
				this.InitializeComponent();
				this.dataRecived.Text = DateTime.Now.ToString();
			}

			private async Task readAsync(SerialDevice serialPort)
			{
				string data = string.Empty;
				using (var dataReaderObject = new DataReader(serialPort.InputStream))
				{
					try
					{
						dataReaderObject.InputStreamOptions = InputStreamOptions.None;
						var hasData = true;
						while (hasData)
						{

							var bytesRead = await dataReaderObject.LoadAsync(serialPort.DataBits);
							if (bytesRead > 0)
								data += dataReaderObject.ReadString(bytesRead);
							else
								hasData = false;

						}
					}
					catch (Exception ex)
					{
						status.Text = "reading data fail: " + ex.Message;
						closeDevice(serialPort);
					}
				}

				dataRecived.Text = data;
			}

			private async void sendData_Click(object sender, RoutedEventArgs e)
			{

				var serialPort = await getSerialDevice("COM3");

				if (dataToSend.Text.Length != 0)
				{

					using (var dataWriteObject = new DataWriter(serialPort.OutputStream))
					{
						try
						{
							if (dataToSend.Text.Equals("#", StringComparison.CurrentCultureIgnoreCase))
								dataWriteObject.WriteString(dataToSend.Text);
							else
								dataWriteObject.WriteString(dataToSend.Text + "\n");

							var bytesWritten = await dataWriteObject.StoreAsync();
							if (bytesWritten > 0)
							{
								status.Text = dataToSend.Text + ", ";
								status.Text += "bytes written successfully!";
							}
							await readAsync(serialPort);

						}
						catch (Exception ex)
						{
							status.Text = "writing data fail: " + ex.Message;
						}
					}
					closeDevice(serialPort);
				}
				else
				{
					status.Text = "Enter the text you want to write and then click on 'WRITE'";
				}
			}

			private void closeDevice(SerialDevice device)
			{
				device.Dispose();
				device = null;
			}

			private async Task<SerialDevice> getSerialDevice(string portName)
			{
				var aqs = SerialDevice.GetDeviceSelector("COM3");
				var devices = await DeviceInformation.FindAllAsync(aqs);

				if (devices.Count == 0)
					throw new Exception("Unablet to connect to device");

				var serialPort = await SerialDevice.FromIdAsync(devices[0].Id);

				if (serialPort == null)
					throw new Exception("Unablet to Open to port " + portName);

				serialPort.BaudRate = 115200;
				serialPort.DataBits = 8;

				return serialPort;
			}

		}
	}
[image]

bfjelds, 2016-01-13:
If this is implemented like other async stream reads, the hasdata=false code will only be executed when the stream is closed.  Otherwise, the LoadAsync will only be triggered when there is data to read.  

You may need to interpret the data after each LoadAsync call is triggered and determine in your code if what you've loaded is complete.

SonOfSam, 2016-01-13:
Basically LoadAsync will in a waiting state if I call it after the last byte read. Can explain when you said create a trigger? you mean look something on the data that come back and that trigger the stop or some other class method to validate?

bfjelds, 2016-01-13:
What I meant was that after each call to LoadAsync returns, check to see if the data that has been received is "complete".  If it is, do not call LoadAsync again.  If it is not, then call LoadAsync again.

IoTGirl, 2016-06-06:
Closing as stale - No response for more than one Month

closed 2016-06-06 by IoTGirl