ms-iot/samples #199: C# and VB Background Task project missing reference to .NET Core

dotMorten, 2016-01-13:
When you download the IoT Core Project templates and create a new background task, the project won't compile. You need to open up project.json and add a reference to .NET Core.

[image]
```
The dependencies entry is missing NETCore 5. When you add this, the project will compile again.
"dependencies": {  
  +		"Microsoft.NETCore.UniversalWindowsPlatform": "5.0.0"  
},  
```

Steps to reproduce:
- Download and install the IoT Core Project Templates from the VS Gallery
- Create a new C# `Background Application (IoT)` project.
- Compile and observe the build errors.

bfjelds, 2016-01-20:
The dependency should be added (as it is for the other C#/VB UWP projects) when the template is instantiated.  

Can you look at the versions you have installed in About Visual Studio?

bfjelds, 2016-02-03:
Please reopen this if you continue seeing this issue!

closed 2016-02-03 by bfjelds