ms-iot/samples #206: App2App WebServer - server crash

MfCAYO, 2016-01-26:
I verified that the Web Server breaks down after 200 http requests and it stops responding.
If you compile the projects in Release config. it breaks too but lasts three times.
(IoT Core version 10.0.10586.63)
Have you got any solutions?
thanks

bfjelds, 2016-02-03:
Your description sounds like it could be related to this issue:

https://github.com/ms-iot/samples/issues/189

bfjelds, 2016-02-03:
Thank you for letting us know about this.  We will investigate.

MfCAYO, 2016-02-03:
ok, thank you for the support!
if I change the following asynchronous methods to synchronous, the stability improves, but I do not think that is the best way...
`// await resp.WriteAsync(headerArray, 0, headerArray.Length);
// await stream.CopyToAsync(resp);
// await resp.FlushAsync();
resp.Write(headerArray, 0, headerArray.Length);
stream.CopyTo(resp);
resp.Flush();`

and than i put the call of ProcessRequestAsync inside a lock block:
` private void listener_ConnectionReceived(StreamSocketListener sender, StreamSocketListenerConnectionReceivedEventArgs args)
        {
            lock (syncLock)
            {
                               ProcessRequestAsync(args.Socket);
            }
        }`

bfjelds, 2016-02-03:
What device are you running into this on?

MfCAYO, 2016-02-03:
Raspberry PI 2

bfjelds, 2016-02-03:
I see instability there too.  I'll see if I can't figure out what is causing this to see if there is a code fix (hopefully!) or there is an issue in the underlying platform.

MfCAYO, 2016-02-03:
thank you very much, I long to see you! ;-)

MfCAYO, 2016-02-21:
hello bfjelds,
sorry but the last published version (Feb 6, 2016) of BlinkyWebService does not start HttpServer and if I try to manually run HttpServer from Windows Device Portal displays the error "Failed to start package HttpServer...".
What can I do?

bfjelds, 2016-02-22:
It is true that the HttpServer needs to be started separately from BlinkyWebService.  It is also unfortunately true that, as a Background Application, it cannot be started from the device portal.

Both should be invoked from Visual Studio using F5.  Please see the sample's deployment documentation here: http://ms-iot.github.io/content/en-US/win10/samples/BlinkyWebServer.htm.

MfCAYO, 2016-02-23:
There is a way for you to use this sample as a default app?
thanks for your invaluable help

bfjelds, 2016-02-23:
For default/startup apps, you can specify one GUI app and any number of Background Applications.  The easiest way to do this would be to:

1. Deploy these using Visual Studio's F5.
2. Use iotstartup (using Powershell or SSH or from the Device Portal's Processes tab):
    
    iotstartup add headed BlinkyWebService
    iotstartup add headless HttpServer

This will configure the OS to start both BlinkyWebService and HttpServer when the system boots up (and make every effort to keep them running should one exit for any reason).  

Keep in mind that the sample was not written to be started on boot and may need to be adjusted (for example, the App-to-app communication currently just tries once to connect ... you may need to try multiple times as the timing may be such that the Background Application starts first ... you may need to watch for any disconnection between the two apps to revive communications ... etc).

closed 2016-09-14 by bfjelds

MfCAYO, 2016-09-14:
thanks again, I applied your advice and usability and reliability are now excellent!