ms-iot/samples #208: BluetoothGATT example for the TI-CC2650

mce1978, 2016-02-04:
The BluetoothGATT example uses the old sensortag CC2541. The CC2541 is no longer available from Texas Instrument and is "replaced" by CC2650.

Shouldn't the sample code be adapted to the new version or that it could handle both versions. The UUID's have changed as well as the formulas to calculate for example temperature.

asebak, 2016-03-24:
Any updates on this?

stephenhuang, 2016-06-15:
Hi guys,
I've passed this on to my team. We'll be looking at revamping some of the samples (including this one) at the end of the month. I'll keep you updated

stephenhuang, 2016-09-07:
Hi guys, after consulting with the team, we'd like to put out an open request to change the sample to not depend on a specific 3rd party device - our focus is less on the sensortag and more on Bluetooth

mce1978, 2016-09-21:
From your point of view I can understand why you want to do that. That's fine! However, this "popular" device has barely got any "good" C# code out there. We, and our customer, wanted to try this device together with the Windows 10 raspberry pie for a proof of concept industry thing. We have dropped that idea, seems like it is easier to go the linux route.

IoTGirl, 2016-12-05:
Hi MCE1978,
I have re-consulted the team and again, they would like to focus on getting generic support rather than any specific device.  I am closing this request as rejected but a third party could certainly tackle any third party hardware support.
Sincerely,
IoTGirl

closed 2016-12-05 by IoTGirl

phardesty, 2017-07-07:
Hello -
Does anyone know the new UUIDs for the new CC2650 Texas Instruments sensorTag that replaced the CC2541. The CC2541 is no longer available from Texas Instrument and was "replaced" by CC2650. The code in the sample uses the old UUID's.

mce1978, 2017-07-12:
The uuids are available at Texas Instruments webpage.

This page is also very informative:
http://processors.wiki.ti.com/index.php/CC2650_SensorTag_User%27s_Guide