ms-iot/samples #210: AllJoyn DSB Control panel

koapeadu, 2016-02-08:
I would like to know if the ControlPanel code files in the AllJoyn DSB project implement both controller and controllee or only the controllee aspect.

zhuridartem, 2016-02-08:
Controlee only. DSB is the application that advertises the control panel for each device that it represents on AllJoyn

koapeadu, 2016-02-10:
Oh wow. Well I understand that. The DSB is meant to enable non-alljoyn devices to be able to communicate in the alljoyn ecosystem so it makes sense to have only the controllee.
Anyway, I would like to know if there is an implementation of the controller side of the control panel service in the UWP or a third party/sample implementation that is compatible with the UWP.

ManniAT, 2016-02-19:
In general Control Panel is put in second place
There are no samples (neither controller nor controllee).
Documentation just shows "...and interface for..." - but no hints (snippets) of how to handle it.

Or am I wrong and just didn't find the information?

zhuridartem, 2016-03-11:
You can find documentation about Control Panel at Allseen Alliance: https://identity.allseenalliance.org/developers/learn/base-services/controlpanel

It links to Samples page
https://identity.allseenalliance.org/developers/develop/run-sample-apps/controlpanel

Note that there are no Windows samples. Code for Controller is available for other platforms on Allseen Alliance.  

It is possible to implement the Controller and Controlee using AllJoyn Studio as a start. The trick with consuming Control Panels is that you can't rely on About Announcements to get the full tree of control panel containers and properties. The Control Panel Interface specifies that you need to use https://dbus.freedesktop.org/doc/dbus-specification.html#standard-interfaces-properties org.freedesktop.DBus.Properties.GetAll() to get the list of Properties in a container. You can't rely on the org.alljoyn.About.GetObjectDescription(), and in our experience anything lower than ControlPanel object was not advertised.

Some more useful information here: https://identity.allseenalliance.org/developers/develop/api-guide/controlpanel

zhuridartem, 2016-03-30:
Closing as Answered.

closed 2016-03-30 by zhuridartem