ms-iot/samples #214: Serial Sample

0kkult0, 2016-02-13:
Hello, I need to write an application for Windows iot, I would use a timer, which listens on com port and is waiting for data sent from the coin selectors, but I can not read the data.

Thanks

Sumahitha, 2016-02-22:
Hello,
Can you provide some more details? Are you using the on-board UART or a USB-to-TTL device? What IoT device are you using? Are you seeing this problem when you use the sample?

Thanks!

0kkult0, 2016-02-22:
Hello, I'm using UART, the device is a coin selectors, I started from the example SerialSample, but I can not read from the coin input.

thanks

Sumahitha, 2016-02-22:
Can you please post the code you are using? Also please give us a brief description of what your expected outcome is -  how are you using the coin device? Are you are using Raspberry Pi or Minnowboard Max or Dragon Board.

0kkult0, 2016-02-23:
I'm only using raspberry pi 2, IOT windows operating system, this is a piece of code that I use to read data from the coin selectors. The myTimer function sends a command to the coin and should intercept a response, but does not work.

private SerialDevice serialPort = null;
DataWriter dataWriteObject = null;
DataReader dataReaderObject = null;
private ObservableCollection<DeviceInformation> listOfDevices;
private CancellationTokenSource ReadCancellationTokenSource;
DispatcherTimer mytimer = new DispatcherTimer();

public MainPage()
{
     this.InitializeComponent();            
     comPortInput.IsEnabled = false;
     sendTextButton.IsEnabled = false;
     listOfDevices = new ObservableCollection<DeviceInformation>();
     ListAvailablePorts();
     OpenPort();
 }

  

        private async void OpenPort()
        {
            ConnectDevices.SelectedIndex = 0;
            var selection = ConnectDevices.SelectedItems;
            DeviceInformation entry = (DeviceInformation)selection[0];
            try
            {
                serialPort = await SerialDevice.FromIdAsync(entry.Id);
                comPortInput.IsEnabled = false;
                serialPort.WriteTimeout = TimeSpan.FromMilliseconds(1000);
                serialPort.ReadTimeout = TimeSpan.FromMilliseconds(1000);
                serialPort.BaudRate = 9600;
                serialPort.Parity = SerialParity.None;
                serialPort.StopBits = SerialStopBitCount.One;
                serialPort.DataBits = 8;
                serialPort.Handshake = SerialHandshake.None;
                status.Text = "Serial port configured successfully: ";
                status.Text += serialPort.BaudRate + "-";
                status.Text += serialPort.DataBits + "-";
                status.Text += serialPort.Parity.ToString() + "-";
                status.Text += serialPort.StopBits;
                rcvdText.Text = "Waiting for data...";
                ReadCancellationTokenSource = new CancellationTokenSource();
                sendTextButton.IsEnabled = true;
                Listen();
            }
            catch (Exception ee) {  }
            
            }

        private async Task WriteAsync(string comando)
        {
            Task<UInt32> storeAsyncTask;
            dataWriteObject.WriteString(comando);                

            // Launch an async task to complete the write operation
            storeAsyncTask = dataWriteObject.StoreAsync().AsTask();

            UInt32 bytesWritten = await storeAsyncTask;
            if (bytesWritten > 0)
            {
                if (status.Text.Length > 200) { status.Text = ""; }
                status.Text += "bytes written successfully!";
            }
           
        }

  
        private async void Listen()
[... 85 lines cut ...]
                    ReadCancellationTokenSource.Cancel();
                }
            }         
        }

        private void CloseDevice()
        {            
            if (serialPort != null)
            {
                serialPort.Dispose();
            }
            serialPort = null;

            comPortInput.IsEnabled = true;
            sendTextButton.IsEnabled = false;            
            rcvdText.Text = "";
            listOfDevices.Clear();               
        }

private void Mytimer_Tick(object sender, object e)
{
     SendE5();
}

`

Sumahitha, 2016-03-02:
The Read operations look correct. Please check the following:
1. Hardware connections -  the TX from the coin device is connected to the RX on the UART pins and RX from the coin device is connected to the TX. Please ensure that GND is also connected correctly
2. You have used DeviceInformation entry = (DeviceInformation)selection[0]. Please make sure that entry [0] corresponds to the UART you have connected to the COIN device. I would suggest to use UART0 (which does not have the control pins). But the coin manual should be able to give better guidance here
3. Also try to simplify your read write to verify and then add on the timer code
4. If possible connect the coin device to your computer via any terminal to verify that the serial port on that device works as expected.
Hope this helps and good luck!

Sumahitha, 2016-03-02:
Also please make sure that the serial capability is added to your project - package manifest

<Capabilities>
    <DeviceCapability Name="serialcommunication">
      <Device Id="any">
        <Function Type="name:serialPort" />
      </Device>
    </DeviceCapability>
  </Capabilities>

Refer here for more details: http://ms-iot.github.io/content/en-US/win10/samples/SerialSample.htm

IoTGirl, 2016-06-03:
Closing - No response from the original poster for more than 1 month

closed 2016-06-03 by IoTGirl