ms-iot/samples #223: string device1 always null

MikaVirkkunen, 2016-02-27:
I'm trying to get this BTSample sample working but it seems that I don't get any values to my "device1" and therefore the device connection always fails.

When I set a break to this line:
string device1 = RfcommDeviceService.GetDeviceSelector(RfcommServiceId.SerialPort);

I can see this:

RfcommServiceId.SerialPort	{Windows.Devices.Bluetooth.Rfcomm.RfcommServiceId}	Windows.Devices.Bluetooth.Rfcomm.RfcommServiceId

device1	null	string

this	{BTSerial.MainPage}	BTSerial.MainPage

What am I missing?

MagicBunny, 2016-03-07:
You need to pair your BT device with the board first before you try the sample.
Can you take a look at this sample steps on how to pair the device first?
https://github.com/ms-iot/content/blob/develop/en-US/win10/samples/BTSerial.md

MikaVirkkunen, 2016-03-07:
Yep it's paired but not working. I've tried with other paired devices too. The result is same if I try to run it locally in my laptop win10. It seems that I get some AMQ response with UUID value etc but device1 seems to be always null. I've also verified the app manifest file values etc and everything seems to be ok. I'm probably missing something really simple here :) in what format the bt device name should be defined? simply the display name which is visible under bluetooth devices or something else like the UUID value?

IoTGirl, 2016-08-01:
Hi Mika,

there was a big push done for connectivity work on RPi2&3 for builds 14366 and higher.  Are you still seeing this issue? If so can you tell us more about your environment?

Sincerely,
IoTGirl

IoTGirl, 2016-10-18:
Closing based on no response after more than a month

closed 2016-10-18 by IoTGirl