ms-iot/samples #226: Windows Universal Runtime version 10.0.10545.0 ?

grahamehorner, 2016-03-04:
I'm unable to run the samples on Windows 10 using VS2015 in a simulated mode;

Error : DEP3321 : To deploy this application, your deployment target should be running Windows Universal Runtime version 10.0.10545.0 or higher. You currently are running version 10.0.10240.16683. Please update your OS, or change your deployment target to a device with the appropriate version.
	IoTCoreDefaultApp			

The system and VisualStudio 2015 are up to date; as of 4th March 2016

bfjelds, 2016-03-07:
What do you mean by "in a simulated mode"?  Also, when you look at **Help > About Visual Studio**, what version of **Visual Studio Tools for Universal Windows Apps** do you have installed?

IoTGirl, 2016-06-03:
Closing: No response from poster in more than one month

closed 2016-06-03 by IoTGirl